TixiaoShan/LIO-SAM #383: [pcl::VoxelGrid::applyFilter] Invalid filter field name. Index is -1.

hichamhendy, 2022-10-11:
I am getting the error when I play records on my jetson whereas when I play  where else every thing is fine

`[pcl::VoxelGrid::applyFilter] Invalid filter field name. Index is -1.
`

withnoerr, 2024-03-13:
> I am getting the error when I play records on my jetson whereas when I play where else every thing is fine
> 
> `[pcl::VoxelGrid::applyFilter] Invalid filter field name. Index is -1. `

请问您解决了吗,我也遇到了同样的错误

hichamhendy, 2024-03-13:
The problem is related to GTSAM version

withnoerr, 2024-03-13:
> The problem is related to GTSAM version

您好,我运行其他slam算法时出现了这个错误,所以这个错误是因为其他依赖库版本不兼容吗,期待您的回复

hichamhendy, 2024-03-13:
Don't understand

withnoerr, 2024-03-13:
我在运行ORB_SLAM2时出现错误 [pcl::VoxelGrid::applyFilter] Invalid filter field name. Index is -1.  无法显示点云,请问我应该如何排查错误原因,网上没有相关错误的解决办法

hichamhendy, 2024-03-13:
Still no Chinese/Japanese