TixiaoShan/LIO-SAM #386: Quaternion Not normalized properly

Dhruv0208, 2022-11-08:
Hi, I was testing lio-sam over my data and I am getting the following errors.
[image]
Any idea where I might be going wrong?

closed 2022-11-18 by Dhruv0208

joao-aguizo, 2022-11-23:
@Dhruv0208 What was the problem? Did you manage to solve it?

buaaysh, 2023-06-30:
Have you solved this problem yet？

buaaysh, 2023-06-30:
> @Dhruv0208 What was the problem? Did you manage to solve it?

Have you solved this problem yet

helywin, 2023-11-27:
@buaaysh I solved this by normalizing IMU data

```cpp
void normalizeImu(sensor_msgs::Imu& imu_msg) {
    double magnitude = std::sqrt(
        imu_msg.orientation.x * imu_msg.orientation.x +
        imu_msg.orientation.y * imu_msg.orientation.y +
        imu_msg.orientation.z * imu_msg.orientation.z +
        imu_msg.orientation.w * imu_msg.orientation.w
    );

    imu_msg.orientation.x /= magnitude;
    imu_msg.orientation.y /= magnitude;
    imu_msg.orientation.z /= magnitude;
    imu_msg.orientation.w /= magnitude;
}
```