TixiaoShan/LIO-SAM #387: ros2 version demo, how to set the config/params.yaml file and which bag to use?

lucianzhong, 2022-11-19:
Hi:
 I am trying to run the ros2 versin of lio-sam, could someone tells how to set the config/params.yaml file and which bag to use?
Thanks a lot

grischi, 2023-01-12:
Hi,

despite minor differences in syntax, configuration shall be similar to the ros1 version, so please follow the recommendations in README.md. You can even use the bags provided for ros1 and bridge to ros2 using https://github.com/ros2/ros1_bridge.

lucianzhong, 2023-01-17:
Hi, thanks for  the replay. Which bag should I use for the default parameters in branch ros2?

grischi, 2023-01-17:
I'm not aware of a dataset that fits the current default parameters. They fit my rosbags, but unfortunately I'm not allowed to publish them for privacy reasons. I therefore suggest to start with the Rooftop Dataset (see README.md) and configure params.yaml as it was before this commit was merged into master: https://github.com/TixiaoShan/LIO-SAM/commit/df05f8a26e52c70240a809837a848ddfee2cfcfb (This commit was not yet merged to ros2 branch).

lucianzhong, 2023-01-18:
Get it. I will try then. Thanks a lot.