TixiaoShan/LIO-SAM #389: About the groundtruth of datasets

MoonSheepJx, 2022-11-21:
First, thank you very much for the excellent work!
i am testing related algorithms on the datasets as you mentioned, such as:
Walking dataset
Park dataset
Garden dataset
but, i check the relative issues, the topic published by `park.bag` and the Google Drive sharing, did not find any groundtruth information.
The groundtruth of datasets is important for quantitative analysis, could you provide some information about your datasets?
Thank you again for your work!
Thanks for any help!

yinhaosen, 2023-03-27:
I think the author used GPS data for ground truth.

dxdydz3dt, 2023-06-24:
Did you get the ground truth? please it with me.
dxdydz3dt@gmail.com