TixiaoShan/LIO-SAM #395: Questions about export each frame point cloud

ooooohhhhhxxx, 2022-12-29:
Your project is great!

I'm new to slam and point cloud processing, and I'm also new to C++. In my project, I am using point clouds from KITTI Raw data, and after SLAM, I need to do other processing on the point clouds for each frame.

I have successfully exported the whole map .pcd file and by modifying the code I have exported the key frame pcd. but as I said I need the point cloud of all the frames and at the moment I am stuck here, if you can help me or give me some advice I would appreciate it!

For example, in the 2011_09_26_drive_0084 data I'm currently using, the original point cloud has 383 frames, but the exported keyframes only have 142 frames.

pauldeee, 2022-12-29:
One quick thing you could try is changing the following settings in the params.yaml to very small values, such as 0.01. This should add keyframes much more frequently.

```
surroundingkeyframeAddingDistThreshold: 1.0   # meters, regulate keyframe adding threshold
surroundingkeyframeAddingAngleThreshold: 0.2  # radians, regulate keyframe adding threshold
```

ooooohhhhhxxx, 2022-12-30:
> 

This method did not work, and even I changed them to 0.001, I think the reason is that in void scan2MapOptimization() in mapOptmization.cpp, it will determine if there are enough of the same feature points in the point cloud before and after the two frames, and if not, it will skip the frame, which should cause the final point cloud frame to be much smaller than the raw point cloud frame.

ooooohhhhhxxx, 2022-12-30:
> One quick thing you could try is changing the following settings in the params.yaml to very small values, such as 0.01. This should add keyframes much more frequently.
>
> ``` [...]

This method did not work, and even I changed them to 0.001, I think the reason is that in void scan2MapOptimization() in mapOptmization.cpp, it will determine if there are enough of the same feature points in the point cloud before and after the two frames, and if not, it will skip the frame, which should cause the final point cloud frame to be much smaller than the raw point cloud frame.

ooooohhhhhxxx, 2022-12-30:
> One quick thing you could try is changing the following settings in the params.yaml to very small values, such as 0.01. This should add keyframes much more frequently.
>
> ``` [...]

But I found another solution, by changing mappingProcessInterval parameter to 0.001, now it can mapping more frequently, for example it now can save 209 frames. I will keep trying.

ooooohhhhhxxx, 2022-12-30:
> But I found another solution, by changing mappingProcessInterval parameter to 0.001, now it can mapping more frequently, for example it now can save 209 frames. I will keep trying.

Here's my final solution, by setting these two parameter as @pauldeee metioned into very small value, also I settingmappingProcessInterval into small value(0.001), most important is slow down the bag play rate, to allow the program to better accept each frame of data, especially when running on virtual machines or computers with poor performance. For example, I use rosbag play kitti_2011_09_26_drive_0084_synced.bag -r 0.1 -d 10 command, to allow bag play at a slow rate of 0.1x. Now I can save all 383 frames. 

Thank you so much @pauldeee , you've given me great ideas. Happy Friday and happy new year!

504115681, 2024-07-19:
> Here's my final solution, by setting these two parameter as @pauldeee metioned into very small value, also I settingmappingProcessInterval into small value(0.001), most important is slow down the [...]

Thank you very much! mappingProcessInterval: 0.15 -> 0.1