TixiaoShan/LIO-SAM #397: "Large velocity, reset IMU-preintegration!"

wwdlh, 2023-01-05:
I tried to run this algorithm with **livox mid70** lidar. The error "_Large velocity, reset IMU preintegration!_" appeared, and the trajectory could be seen flying in _rviz_. The parameters related to the lidar are set according to the Livox Horizon lidar in README.md. The livox_ros_driver used is also the version that can publish the correct message type after the author modifying it. The imu I used is **ibc-nav40**. This imu will not make mistakes when running the lio-sam with the velodyne vlp16 lidar. Therefore, I guess it is a problem that the mid70 solid-state laser radar is not supported. I want to ask the author and everyone for help, is there a way to solve this problem by modifying the parameter file or part of the code

Zermelo-101, 2023-03-20:
Did you find a fix to this problem ?

JokerJohn, 2023-04-18:
I believe the main reason for this issue is that the pre-integration node constructs a factor graph of size 100 and uses optimized lidar pose to estimate IMU bias. This factor graph is reset and re-propagated with new bias every time 100 lidar poses are received, causing the predicted pose to be discontinuous and resulting in system crashes. I suggest trying to reduce the weight of lidar pose. 
`(priorPoseNoise = gtsam::noiseModel::Diagonal::Sigmas((gtsam::Vector(6) << 1e-2, 1e-2, 1e-2, 1e-2, 1e-2, 1e-2).finished()); // rad,rad,rad,m, m, m)`

Stream11, 2023-10-08:
@JokerJohn Hello, I encountered this error when I was running a very large package (greater than 1000s). .The front part of the mapping is normal. Do you have any suggestions? Thank you