TixiaoShan/LIO-SAM #398: Compile LIO-SAM on Ubuntu 22.04

fupenghzau, 2023-01-16:
I am trying to compile LIO-SAM on Ubuntu 22.04 with ROS2. Below is the error I got after the command 

colcon build

include/lio_sam/utility.hpp:16:10: fatal error: visualization_msgs/msg/marker.hpp: No such file or directory
   16 | #include <visualization_msgs/msg/marker.hpp>
pcl_conversions.h:61:10: fatal error: pcl_msgs/msg/point_indices.hpp: No such file or directory
   61 | #include <pcl_msgs/msg/point_indices.hpp>
compilation terminated.
gmake[2]: *** [CMakeFiles/lio_sam_mapOptimization.dir/build.make:76: CMakeFiles/lio_sam_mapOptimization.dir/src/mapOptmization.cpp.o] Error 1
gmake[1]: *** [CMakeFiles/Makefile2:684: CMakeFiles/lio_sam_mapOptimization.dir/all] Error 2
gmake: *** [Makefile:146: all] Error 2

Anyone knows what are the issues here?

fupenghzau, 2023-01-16:
I can find these header files: marker.hpp, point_indces.hpp in the /opt/ros/humble/include folder. But somehow the compile still runs wrong. I guess it is some lines that I need to add to Cmakelists but not sure how to do it. Any suggestions?

grischi, 2023-01-17:
Suggestion: 1. remove build folder 2. open new terminal 3. verify you have all packages installed as mentioned in `README.md` 4. `source `the humble `setup.bash` from `/opt` 5. change to your ros2 workspace and `colcon build`
Note: you will have to build gtsam from source for Ubuntu 22.04.

grischi, 2023-01-17:
This problem is caused by missing dependencies in CMakeLists.txt, I'm preparing a fix.

fupenghzau, 2023-01-17:
Thanks. Looking forward to it.

fupenghzau, 2023-01-18:
It works. Thanks!
Below is the warning I got from the compilation. Just want to share in case there is any interest.
--- stderr: lio_sam                                          
** WARNING ** io features related to pcap will be disabled
CMake Deprecation Warning at /opt/ros/humble/share/rosidl_cmake/cmake/rosidl_target_interfaces.cmake:32 (message):
  Use rosidl_get_typesupport_target() and target_link_libraries() instead of
  rosidl_target_interfaces()
Call Stack (most recent call first):
  CMakeLists.txt:38 (rosidl_target_interfaces)

CMake Deprecation Warning at /opt/ros/humble/share/rosidl_cmake/cmake/rosidl_target_interfaces.cmake:32 (message):
  Use rosidl_get_typesupport_target() and target_link_libraries() instead of
  rosidl_target_interfaces()
Call Stack (most recent call first):
  CMakeLists.txt:42 (rosidl_target_interfaces)

CMake Deprecation Warning at /opt/ros/humble/share/rosidl_cmake/cmake/rosidl_target_interfaces.cmake:32 (message):
  Use rosidl_get_typesupport_target() and target_link_libraries() instead of
  rosidl_target_interfaces()
Call Stack (most recent call first):
  CMakeLists.txt:47 (rosidl_target_interfaces)

CMake Deprecation Warning at /opt/ros/humble/share/rosidl_cmake/cmake/rosidl_target_interfaces.cmake:32 (message):
  Use rosidl_get_typesupport_target() and target_link_libraries() instead of
  rosidl_target_interfaces()
Call Stack (most recent call first):
  CMakeLists.txt:52 (rosidl_target_interfaces)

/usr/bin/ld: warning: libtbb.so.2, needed by /usr/lib/x86_64-linux-gnu/libopencv_core.so.4.5.4d, may conflict with libtbb.so.12
---
Finished <<< lio_sam [1min 42s]

fupenghzau, 2023-01-18:
After compilation, when I launch it I got the following error:

[INFO] [launch]: All log files can be found below /home/pengfu/.ros/log/2023-01-18-17-54-24-062725-thor-145245
[INFO] [launch]: Default logging verbosity is set to INFO
urdf_file_name : /home/pengfu/lidar/ros2_ws/src/install/lio_sam/share/lio_sam/config/robot.urdf.xacro
[ERROR] [launch]: Caught exception in launch (see debug for traceback): file not found: [Errno 2] No such file or directory: 'xacro'

Do you know why I am missing xacro file? I did find the /home/pengfu/lidar/ros2_ws/src/install/lio_sam/share/lio_sam/config/robot.urdf.xacro

grischi, 2023-01-19:
> ## /usr/bin/ld: warning: libtbb.so.2, needed by /usr/lib/x86_64-linux-gnu/libopencv_core.so.4.5.4d, may conflict with libtbb.so.12

Thanks for the hint, I pushed a fix in https://github.com/TixiaoShan/LIO-SAM/commit/06da1cc5d0a3228a1721382bfb67d0ca45ab3b4a

grischi, 2023-01-19:
> After compilation, when I launch it I got the following error:
>
> [INFO] [launch]: All log files can be found below /home/pengfu/.ros/log/2023-01-18-17-54-24-062725-thor-145245 [INFO] [launch]: [...]

I suppose you're missing the xacro binary. Have you installed ros-humble-xacro as mentioned in README.md?

fupenghzau, 2023-01-19:
> I suppose you're missing the xacro binary. Have you installed ros-humble-xacro as mentioned in README.md?

Yes, I did not install xacro and got it fixed.

closed 2023-01-24 by grischi