TixiaoShan/LIO-SAM #400: Compiling LIO-SAM on Ubuntu 22.04 and ROS2

fferroni, 2023-02-01:
Hello, 

I tried to follow the compilation instructions on the ROS2 branch, but got the following error. Any tips would be much appreciated!

```
(base) francesco@razer:~/ros2_ws/src$ colcon build
Starting >>> lio_sam 
[Processing: lio_sam]                             
--- stderr: lio_sam                               
Traceback (most recent call last):
  File "/opt/ros/humble/share/rosidl_generator_py/cmake/../../../lib/rosidl_generator_py/rosidl_generator_py", line 8, in <module>
    from rosidl_generator_py import generate_py
ImportError: cannot import name 'generate_py' from 'rosidl_generator_py' (/opt/ros/humble/local/lib/python3.10/dist-packages/rosidl_generator_py/__init__.py)
gmake[2]: *** [lio_sam__py/CMakeFiles/lio_sam__py.dir/build.make:184: rosidl_generator_py/lio_sam/_lio_sam_s.ep.rosidl_typesupport_fastrtps_c.c] Error 1
gmake[1]: *** [CMakeFiles/Makefile2:710: lio_sam__py/CMakeFiles/lio_sam__py.dir/all] Error 2
gmake[1]: *** Waiting for unfinished jobs....
In file included from /home/francesco/ros2_ws/src/LIO-SAM/src/mapOptmization.cpp:9:
/usr/include/gtsam/navigation/ImuFactor.h: In member function ‘void gtsam::ImuFactor2::serialize(ARCHIVE&, unsigned int)’:
/usr/include/gtsam/navigation/ImuFactor.h:328:32: error: ‘make_nvp’ is not a member of ‘boost::serialization’
  328 |     ar & boost::serialization::make_nvp("NoiseModelFactor3",
      |                                ^~~~~~~~
/usr/include/gtsam/navigation/ImuFactor.h:329:32: error: ‘base_object’ is not a member of ‘boost::serialization’
  329 |          boost::serialization::base_object<Base>(*this));
      |                                ^~~~~~~~~~~
/usr/include/gtsam/navigation/ImuFactor.h:329:48: error: expected primary-expression before ‘>’ token
  329 |          boost::serialization::base_object<Base>(*this));
      |                                                ^
/usr/include/gtsam/navigation/ImuFactor.h:330:10: error: there are no arguments to ‘BOOST_SERIALIZATION_NVP’ that depend on a template parameter, so a declaration of ‘BOOST_SERIALIZATION_NVP’ must be available [-fpermissive]
  330 |     ar & BOOST_SERIALIZATION_NVP(_PIM_);
      |          ^~~~~~~~~~~~~~~~~~~~~~~
/usr/include/gtsam/navigation/ImuFactor.h:330:10: note: (if you use ‘-fpermissive’, G++ will accept your code, but allowing the use of an undeclared name is deprecated)
In file included from /home/francesco/ros2_ws/src/LIO-SAM/src/imuPreintegration.cpp:8:
/usr/include/gtsam/navigation/ImuFactor.h: In member function ‘void gtsam::ImuFactor2::serialize(ARCHIVE&, unsigned int)’:
/usr/include/gtsam/navigation/ImuFactor.h:328:32: error: ‘make_nvp’ is not a member of ‘boost::serialization’
  328 |     ar & boost::serialization::make_nvp("NoiseModelFactor3",
      |                                ^~~~~~~~
/usr/include/gtsam/navigation/ImuFactor.h:329:32: error: ‘base_object’ is not a member of ‘boost::serialization’
  329 |          boost::serialization::base_object<Base>(*this));
      |                                ^~~~~~~~~~~
/usr/include/gtsam/navigation/ImuFactor.h:329:48: error: expected primary-expression before ‘>’ token
  329 |          boost::serialization::base_object<Base>(*this));
      |                                                ^
/usr/include/gtsam/navigation/ImuFactor.h:330:10: error: there are no arguments to ‘BOOST_SERIALIZATION_NVP’ that depend on a template parameter, so a declaration of ‘BOOST_SERIALIZATION_NVP’ must be available [-fpermissive]
  330 |     ar & BOOST_SERIALIZATION_NVP(_PIM_);
      |          ^~~~~~~~~~~~~~~~~~~~~~~
/usr/include/gtsam/navigation/ImuFactor.h:330:10: note: (if you use ‘-fpermissive’, G++ will accept your code, but allowing the use of an undeclared name is deprecated)
/home/francesco/ros2_ws/src/LIO-SAM/src/imuPreintegration.cpp: In constructor ‘IMUPreintegration::IMUPreintegration(const rclcpp::NodeOptions&)’:
/home/francesco/ros2_ws/src/LIO-SAM/src/imuPreintegration.cpp:248:100: error: conversion from ‘std::shared_ptr<gtsam::PreintegrationParams>’ to non-scalar type ‘boost::shared_ptr<gtsam::PreintegrationParams>’ requested
  248 |         boost::shared_ptr<gtsam::PreintegrationParams> p = gtsam::PreintegrationParams::MakeSharedU(imuGravity);
      |                                                            ~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~^~~~~~~~~~~~
/home/francesco/ros2_ws/src/LIO-SAM/src/imuPreintegration.cpp:261:86: error: no matching function for call to ‘gtsam::PreintegratedImuMeasurements::PreintegratedImuMeasurements(boost::shared_ptr<gtsam::PreintegrationParams>&, gtsam::imuBias::ConstantBias&)’
  261 |         imuIntegratorImu_ = new gtsam::PreintegratedImuMeasurements(p, prior_imu_bias); // setting up the IMU integration for IMU message thread
      |                                                                                      ^
In file included from /home/francesco/ros2_ws/src/LIO-SAM/src/imuPreintegration.cpp:8:
/usr/include/gtsam/navigation/ImuFactor.h:105:3: note: candidate: ‘gtsam::PreintegratedImuMeasurements::PreintegratedImuMeasurements(const PreintegrationType&, const Matrix9&)’
  105 |   PreintegratedImuMeasurements(const PreintegrationType& base, const Matrix9& preintMeasCov)
      |   ^~~~~~~~~~~~~~~~~~~~~~~~~~~~
/usr/include/gtsam/navigation/ImuFactor.h:105:58: note:   no known conversion for argument 1 from ‘boost::shared_ptr<gtsam::PreintegrationParams>’ to ‘const PreintegrationType&’ {aka ‘const gtsam::TangentPreintegration&’}
  105 |   PreintegratedImuMeasurements(const PreintegrationType& base, const Matrix9& preintMeasCov)
      |                                ~~~~~~~~~~~~~~~~~~~~~~~~~~^~~~
/usr/include/gtsam/navigation/ImuFactor.h:94:3: note: candidate: ‘gtsam::PreintegratedImuMeasurements::PreintegratedImuMeasurements(const std::shared_ptr<gtsam::PreintegrationParams>&, const gtsam::imuBias::ConstantBias&)’
   94 |   PreintegratedImuMeasurements(const std::shared_ptr<PreintegrationParams>& p,
      |   ^~~~~~~~~~~~~~~~~~~~~~~~~~~~
/usr/include/gtsam/navigation/ImuFactor.h:94:77: note:   no known conversion for argument 1 from ‘boost::shared_ptr<gtsam::PreintegrationParams>’ to ‘const std::shared_ptr<gtsam::PreintegrationParams>&’
   94 |   PreintegratedImuMeasurements(const std::shared_ptr<PreintegrationParams>& p,
      |                                ~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~^
/usr/include/gtsam/navigation/ImuFactor.h:85:3: note: candidate: ‘gtsam::PreintegratedImuMeasurements::PreintegratedImuMeasurements()’
   85 |   PreintegratedImuMeasurements() {
      |   ^~~~~~~~~~~~~~~~~~~~~~~~~~~~
/usr/include/gtsam/navigation/ImuFactor.h:85:3: note:   candidate expects 0 arguments, 2 provided
[... 20 lines cut ...]
/usr/include/gtsam/navigation/ImuFactor.h:85:3: note: candidate: ‘gtsam::PreintegratedImuMeasurements::PreintegratedImuMeasurements()’
   85 |   PreintegratedImuMeasurements() {
      |   ^~~~~~~~~~~~~~~~~~~~~~~~~~~~
/usr/include/gtsam/navigation/ImuFactor.h:85:3: note:   candidate expects 0 arguments, 2 provided
/usr/include/gtsam/navigation/ImuFactor.h:72:20: note: candidate: ‘gtsam::PreintegratedImuMeasurements::PreintegratedImuMeasurements(const gtsam::PreintegratedImuMeasurements&)’
   72 | class GTSAM_EXPORT PreintegratedImuMeasurements: public PreintegrationType {
      |                    ^~~~~~~~~~~~~~~~~~~~~~~~~~~~
/usr/include/gtsam/navigation/ImuFactor.h:72:20: note:   candidate expects 1 argument, 2 provided
gmake[2]: *** [CMakeFiles/lio_sam_imuPreintegration.dir/build.make:76: CMakeFiles/lio_sam_imuPreintegration.dir/src/imuPreintegration.cpp.o] Error 1
gmake[1]: *** [CMakeFiles/Makefile2:656: CMakeFiles/lio_sam_imuPreintegration.dir/all] Error 2
gmake[2]: *** [CMakeFiles/lio_sam_mapOptimization.dir/build.make:76: CMakeFiles/lio_sam_mapOptimization.dir/src/mapOptmization.cpp.o] Error 1
gmake[1]: *** [CMakeFiles/Makefile2:684: CMakeFiles/lio_sam_mapOptimization.dir/all] Error 2
gmake: *** [Makefile:146: all] Error 2
---
Failed   <<< lio_sam [44.7s, exited with code 2]

Summary: 0 packages finished [44.9s]
  1 package failed: lio_sam
  1 package had stderr output: lio_sam
```

I have this version of boost
```
libboost-serialization-dev is already the newest version (1.74.0.3ubuntu7).
```

yuma116, 2023-02-02:
I have a same problem on ROS1 of Ubuntu20.04
```
nuc@nuc:~/ros1_ws/liosam01_ws
$ catkin_make
Base path: /home/nuc/ros1_ws/liosam01_ws
Source space: /home/nuc/ros1_ws/liosam01_ws/src
Build space: /home/nuc/ros1_ws/liosam01_ws/build
Devel space: /home/nuc/ros1_ws/liosam01_ws/devel
Install space: /home/nuc/ros1_ws/liosam01_ws/install
####
#### Running command: "make cmake_check_build_system" in "/home/nuc/ros1_ws/liosam01_ws/build"
####
-- Using CATKIN_DEVEL_PREFIX: /home/nuc/ros1_ws/liosam01_ws/devel
-- Using CMAKE_PREFIX_PATH: /opt/ros/noetic
-- This workspace overlays: /opt/ros/noetic
-- Found PythonInterp: /usr/bin/python3 (found suitable version "3.8.10", minimum required is "3") 
-- Using PYTHON_EXECUTABLE: /usr/bin/python3
-- Using Debian Python package layout
-- Using empy: /usr/lib/python3/dist-packages/em.py
-- Using CATKIN_ENABLE_TESTING: ON
-- Call enable_testing()
-- Using CATKIN_TEST_RESULTS_DIR: /home/nuc/ros1_ws/liosam01_ws/build/test_results
-- Forcing gtest/gmock from source, though one was otherwise available.
-- Found gtest sources under '/usr/src/googletest': gtests will be built
-- Found gmock sources under '/usr/src/googletest': gmock will be built
-- Found PythonInterp: /usr/bin/python3 (found version "3.8.10") 
-- Using Python nosetests: /usr/bin/nosetests3
-- catkin 0.8.10
-- BUILD_SHARED_LIBS is on
-- BUILD_SHARED_LIBS is on
-- ~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
-- ~~  traversing 17 packages in topological order:
-- ~~  - ublox (metapackage)
-- ~~  - velodyne (metapackage)
-- ~~  - velodyne_msgs
-- ~~  - nl_tools
-- ~~  - lidar_imu_calib
-- ~~  - microstrain_mips
-- ~~  - lio_sam
-- ~~  - ndt_omp
-- ~~  - robot_localization
-- ~~  - ublox_serialization
-- ~~  - ublox_msgs
-- ~~  - ublox_gps
-- ~~  - ublox_msg_filters
-- ~~  - velodyne_driver
-- ~~  - velodyne_laserscan
-- ~~  - velodyne_pcl
-- ~~  - velodyne_pointcloud
-- ~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
-- +++ processing catkin metapackage: 'ublox'
-- ==> add_subdirectory(lio-sam/ublox/ublox)
-- +++ processing catkin metapackage: 'velodyne'
-- ==> add_subdirectory(lio-sam/velodyne/velodyne)
-- +++ processing catkin package: 'velodyne_msgs'
-- ==> add_subdirectory(lio-sam/velodyne/velodyne_msgs)
-- Using these message generators: gencpp;geneus;genlisp;gennodejs;genpy
-- velodyne_msgs: 2 messages, 0 services
-- +++ processing catkin package: 'nl_tools'
-- ==> add_subdirectory(lio-sam/nl_tools)
-- Using these message generators: gencpp;geneus;genlisp;gennodejs;genpy
-- Eigen found (include: /usr/include/eigen3, version: 3.3.7)
-- Found Boost: /usr/include (found suitable version "1.71.0", minimum required is "1.55.0") found components: system filesystem date_time iostreams regex 
-- The imported target "vtkParseOGLExt" references the file
   "/usr/bin/vtkParseOGLExt-7.1"
but this file does not exist.  Possible reasons include:
* The file was deleted, renamed, or moved to another location.
* An install or uninstall procedure did not complete successfully.
* The installation package was faulty and contained
   "/usr/lib/cmake/vtk-7.1/VTKTargets.cmake"
[... 1037 lines cut ...]
Scanning dependencies of target filter_base
[ 93%] Building CXX object lio-sam/robot_localization/CMakeFiles/filter_base.dir/src/filter_base.cpp.o
[ 93%] Built target get_gyro_bias_client
Scanning dependencies of target navsat_transform
[ 93%] Building CXX object lio-sam/robot_localization/CMakeFiles/navsat_transform.dir/src/navsat_transform.cpp.o
make[2]: *** [lio-sam/LIO-SAM/CMakeFiles/lio_sam_imuPreintegration.dir/build.make:63: lio-sam/LIO-SAM/CMakeFiles/lio_sam_imuPreintegration.dir/src/imuPreintegration.cpp.o] Error 1
make[1]: *** [CMakeFiles/Makefile2:6981: lio-sam/LIO-SAM/CMakeFiles/lio_sam_imuPreintegration.dir/all] Error 2
make[1]: *** Waiting for unfinished jobs....
[ 93%] Linking CXX executable /home/nuc/ros1_ws/liosam01_ws/devel/lib/microstrain_mips/microstrain_mips_node
[ 93%] Linking CXX executable /home/nuc/ros1_ws/liosam01_ws/devel/lib/microstrain_mips/set_soft_iron_matrix_client
[ 93%] Built target microstrain_mips_node
[ 93%] Built target set_soft_iron_matrix_client
[ 94%] Linking CXX shared library /home/nuc/ros1_ws/liosam01_ws/devel/lib/libfilter_base.so
[ 94%] Built target filter_base
[ 95%] Linking CXX executable /home/nuc/ros1_ws/liosam01_ws/devel/lib/lio_sam/lio_sam_featureExtraction
[ 95%] Linking CXX executable /home/nuc/ros1_ws/liosam01_ws/devel/lib/lio_sam/lio_sam_imageProjection
[ 95%] Built target lio_sam_featureExtraction
[ 95%] Built target lio_sam_imageProjection
make[2]: *** [lio-sam/LIO-SAM/CMakeFiles/lio_sam_mapOptmization.dir/build.make:63: lio-sam/LIO-SAM/CMakeFiles/lio_sam_mapOptmization.dir/src/mapOptmization.cpp.o] Error 1
make[1]: *** [CMakeFiles/Makefile2:7069: lio-sam/LIO-SAM/CMakeFiles/lio_sam_mapOptmization.dir/all] Error 2
[ 95%] Linking CXX shared library /home/nuc/ros1_ws/liosam01_ws/devel/lib/libnavsat_transform.so
[ 95%] Built target navsat_transform
make: *** [Makefile:141: all] Error 2
Invoking "make -j8 -l8" failed
```

yuma116, 2023-02-02:
My version is it
```
$ dpkg -l | grep libboost-serialization-dev
ii  libboost-serialization-dev:amd64            1.71.0.0ubuntu2                     amd64        serialization library for C++ (default version)
```

bumblebeeCC, 2023-02-02:
Same issue  on ROS2 branch with ROS2 humble/22.04 LTS, root cause might be the prerequisite [gtsam](https://github.com/borglab/gtsam), ROS2 branch recommend install "develop"  unstable release, but gtsam has removed boost on Jan 2023.

"Important Note

As of January 2023, the develop branch is officially in "Pre 4.3" mode. We envision several API-breaking changes as we switch to C++17 and away from boost. "

grischi, 2023-02-02:
> 

as i see this line:
`ImportError: cannot import name 'generate_py' from 'rosidl_generator_py' (/opt/ros/humble/local/lib/python3.10/dist-packages/rosidl_generator_py/__init__.py)`

have you tried the solution suggested here?: https://answers.ros.org/question/407879/humble-tutorial-creating-a-workspace/

grischi, 2023-02-02:
the other problem with gtsam can (as indicated by @bumblebeeCC ) be resolved by selecting the gtsam release ppa, like so:
`sudo add-apt-repository ppa:borglab/gtsam-release-4.1`

grischi, 2023-02-02:
> Same issue on ROS2 branch with ROS2 humble/22.04 LTS, root cause might be the prerequisite [gtsam](https://github.com/borglab/gtsam), ROS2 branch recommend install "develop" unstable release, but [...]

Note: Using the "develop" ppa was recommended because the release ppas did not provide packages for 22.04 at that time.

fferroni, 2023-02-03:
OK, thanks, that seems to have fixed the boost issue, but I get a new issue (I wiped my workspace and restarted).

```
(base) francesco@razer:~/ros2_ws/src$ colcon build
Starting >>> lio_sam 
[Processing: lio_sam]                             
[Processing: lio_sam]                                     
[Processing: lio_sam]                                       
--- stderr: lio_sam                                         
** WARNING ** io features related to pcap will be disabled
CMake Deprecation Warning at /opt/ros/humble/share/rosidl_cmake/cmake/rosidl_target_interfaces.cmake:32 (message):
  Use rosidl_get_typesupport_target() and target_link_libraries() instead of
  rosidl_target_interfaces()
Call Stack (most recent call first):
  CMakeLists.txt:39 (rosidl_target_interfaces)

CMake Deprecation Warning at /opt/ros/humble/share/rosidl_cmake/cmake/rosidl_target_interfaces.cmake:32 (message):
  Use rosidl_get_typesupport_target() and target_link_libraries() instead of
  rosidl_target_interfaces()
Call Stack (most recent call first):
  CMakeLists.txt:43 (rosidl_target_interfaces)

CMake Deprecation Warning at /opt/ros/humble/share/rosidl_cmake/cmake/rosidl_target_interfaces.cmake:32 (message):
  Use rosidl_get_typesupport_target() and target_link_libraries() instead of
  rosidl_target_interfaces()
Call Stack (most recent call first):
  CMakeLists.txt:48 (rosidl_target_interfaces)

CMake Deprecation Warning at /opt/ros/humble/share/rosidl_cmake/cmake/rosidl_target_interfaces.cmake:32 (message):
  Use rosidl_get_typesupport_target() and target_link_libraries() instead of
  rosidl_target_interfaces()
Call Stack (most recent call first):
  CMakeLists.txt:53 (rosidl_target_interfaces)

Traceback (most recent call last):
  File "/opt/ros/humble/share/rosidl_generator_py/cmake/../../../lib/rosidl_generator_py/rosidl_generator_py", line 8, in <module>
    from rosidl_generator_py import generate_py
ImportError: cannot import name 'generate_py' from 'rosidl_generator_py' (/opt/ros/humble/local/lib/python3.10/dist-packages/rosidl_generator_py/__init__.py)
gmake[2]: *** [lio_sam__py/CMakeFiles/lio_sam__py.dir/build.make:184: rosidl_generator_py/lio_sam/_lio_sam_s.ep.rosidl_typesupport_fastrtps_c.c] Error 1
gmake[1]: *** [CMakeFiles/Makefile2:710: lio_sam__py/CMakeFiles/lio_sam__py.dir/all] Error 2
gmake[1]: *** Waiting for unfinished jobs....
/usr/bin/ld: warning: libtbb.so.2, needed by /usr/lib/x86_64-linux-gnu/libopencv_core.so.4.5.4d, may conflict with libtbb.so.12
gmake: *** [Makefile:146: all] Error 2
---
Failed   <<< lio_sam [1min 50s, exited with code 2]

Summary: 0 packages finished [1min 50s]
  1 package failed: lio_sam
  1 package had stderr output: lio_sam
```

zuowei1998, 2023-05-08:
@yuma116 hello，Did you solve this problem, I have the same problem。 Thanks.

closed 2023-05-09 by grischi