TixiaoShan/LIO-SAM #402: What is the difference between * and *_ incremental?

Jack1412, 2023-02-22:
What is the difference between * and *_ incremental, how to understand it. Thank you!
Below is a test image:
[image]

sevagul, 2023-02-28:
As far as I understand from the code, here is the difference:

Non-incremental odometries are the **best last guess** of the robot's location.
It *may jump* after loop closures and GPS measurement integrations.  <br/>
At the same time, `*_incremental` odometries are **smooth**. Those are unaffected by the global optimization (loop closure, GPS updates), and only reflect the best guess of the last motion. The best guess of the last motion is added to the previous incremental odometry. This type of odometry is widely used when the robot jumps badly affect the algorithm, for instance, during robot control, etc. <br/>

How those two `_incremental` odometries are used in LIO-SAM: <br/>

Incremental laser odometry ( `.../mapping/odometry_incremental` ) is used by the IMU Preintegration node to adequately estimate IMU Bias. In case it would use a last-best-guess `.../mapping/odometry`, the graph would contain sudden jumps, and IMU Bias would be estimated not correctly. Probably, the best way would be to reconstruct that graph every time, using a best-guess path, but I think that would be computationally much heavier. <br/>
Incremental IMU odometry ( `odometry/imu_incremental` ) is used for deskewing in `imageProjection.cpp`. Again, if it would always store the last best guess, jumps of the position would deskew the pointcloud in the wrong way. <br/>

Please correct me if I am wrong. Hopefully, this reply would help.

Nothand0212, 2023-07-04:
> As far as I understand from the code, here is the difference:
> 
> Non-incremental odometries are the **best last guess** of the robot's location. It _may jump_ after loop closures and GPS measurement integrations. At the same time, `*_incremental` odometries are **smooth**. Those are unaffected by the global optimization (loop closure, GPS updates), and only reflect the best guess of the last motion. The best guess of the last motion is added to the previous incremental odometry. This type of odometry is widely used when the robot jumps badly affect the algorithm, for instance, during robot control, etc.
> 
> How those two `_incremental` odometries are used in LIO-SAM:
> 
> Incremental laser odometry ( `.../mapping/odometry_incremental` ) is used by the IMU Preintegration node to adequately estimate IMU Bias. In case it would use a last-best-guess `.../mapping/odometry`, the graph would contain sudden jumps, and IMU Bias would be estimated not correctly. Probably, the best way would be to reconstruct that graph every time, using a best-guess path, but I think that would be computationally much heavier. Incremental IMU odometry ( `odometry/imu_incremental` ) is used for deskewing in `imageProjection.cpp`. Again, if it would always store the last best guess, jumps of the position would deskew the pointcloud in the wrong way.
> 
> Please correct me if I am wrong. Hopefully, this reply would help.

so, if I want to apply LIO-SAM on robot navigation, I should subscribe `odometry_incremental`?