TixiaoShan/LIO-SAM #410: ROS2 save pcd file failed @hitbg-yjm

lushengyao, 2023-03-30:
thanks for your great job, and   how to solve this problem in ROS2

            @hitbg-yjm 

- [I just made a new push for this problem.](https://github.com/TixiaoShan/LIO-SAM/commit/64a10e8e996ffee7154f144dbdeb6e2425dea9b9)
- You also need to use **```sudo gedit /opt/ros/kinetic/lib/python2.7/dist-packages/roslaunch/nodeprocess.py```** to change the timeout threshold for killing the node. ROS will kill the node after **```_TIMEOUT_SIGINT```** seconds. A big map might need a few minutes to save. Change it to a larger value if needed.

_Originally posted by @TixiaoShan in https://github.com/TixiaoShan/LIO-SAM/issues/3#issuecomment-653493972_

Gaoeee, 2023-09-08:
Did you solve it , bro?

Saiteja-RoshAi, 2024-07-09:
@Gaoeee is it solved bro?

wxyLuna2024, 2024-07-18:
Also looking into this problem. When I turned on savePCD, nothing is being saved and stored in "/Downloads/LOAM/". When mapping is finished, I tried to save map manually, it says waiting for service to become available...

Saiteja-RoshAi, 2024-07-19:
> Also looking into this problem. When I turned on savePCD, nothing is being saved and stored in "/Downloads/LOAM/". When mapping is finished, I tried to save map manually, it says waiting for service to become available...

@wxyLuna2024 you need to call the service 
rosservice call resolution "path"
for ex: rosservice call 0.2 "/Download/LOAM/"

it's working for me , wait for sometime depending on the size of the map it takes time to save the map.

nmin0430, 2025-02-28:
I already use ros2 service call /lio_sam/save_map lio_sam/srv/SaveMap "{resolution: 0.2, destination: /Downloads/service_LOAM}
it only show '>' or 'waiting for service to become available...' i check the ros2 service list it have lio_sam/save_map,what can i do next

XX-4869, 2025-04-27:
> I already use ros2 service call /lio_sam/save_map lio_sam/srv/SaveMap "{resolution: 0.2, destination: /Downloads/service_LOAM} it only show '>' or 'waiting for service to become available...' i check the ros2 service list it have lio_sam/save_map,what can i do next

你是怎么解决的，兄弟？