TixiaoShan/LIO-SAM #417: GPS Implementation

mistercherryy, 2023-04-11:
Hi,

I would like to integrate my own GPS data to Lio-sam. The topic names of IMU and Lidar data is different but i can get the trajectory and the map according to those data. The only part left is Integrating Gps but unfortunately "odometry/gps" publishes nothing so far even though i feed a gps topic into navsat transform node. 

The only thing i see missing so far is "imu_correct" in my case. Instead i feed the same "imu_raw" topic that i feed for imu preintegration to Navsat  because i don't have a transformation from my own imu data to the kind of imu_correct.

If anyone has a solution for this issue, i will be appreciated for any help.

Ahmad-Drak, 2023-06-10:
Im in the same place you are. Did you manage to fix it?

JokerJohn, 2023-08-30:
@mistercherryy @Ahmad-Drak I believe this issue is quite straightforward. In LIO-SAM, the GPS factor is fused in a post-fusion manner (loosely coupled), directly using the built-in GPS factor from GTSAM. Regarding the data format, LIO-SAM requires a 9-axis IMU and GPS to be fused through the robot-localization node. This is to provide stable GNSS position observations and IMU initialization. For LIO-SAM, apart from the initialization, the state estimation mainly requires a 6-axis IMU (3-axis acceleration and 3-axis angular velocity data). Based on the above reasons, it's perfectly fine to feed the IMU and GPS separately. Please refer to: [LIO_SAM_6AXIS](https://github.com/JokerJohn/LIO_SAM_6AXIS). The only requirement is to remain relatively still during the initial phase and avoid large-angle rotational movements to ensure IMU initialization.