TixiaoShan/LIO-SAM #422: The point cloud running and shaking are very serious, what went wrong?

ldzhuang, 2023-05-06:
[image]
[image]
[image]

https://user-images.githubusercontent.com/64113233/236616203-377bd025-bc43-4bce-b852-b35db45a3b5b.mp4

ldzhuang, 2023-05-12:
Thank you for your answer, but I have encountered a problem

Iesenjin, 2023-05-12:
From my very limited experience - as I am a beginner both in LIO SAM and programing in general - this may be because you are using wrong matrices. While experimenting with values, sometimes the recording did that, because LIO SAM was confused by the output data from IMU and LIDAR.
Try first matrix set, and see how the recording will behave.
Also - use default values for extrinsicTrans, at least until you stabilize the recording.