TixiaoShan/LIO-SAM #427: IMU preintegration restarted

HomieRegina, 2023-05-16:
Hi, thanks for your great work! I tried lio-sam to build the map. But when the bag is playing, it's normal at first. After a while, the terminal start reporting errors. There is a picture about the error. Have you ever met this error? Do you know how to solve this problem? 
[image]

Look forward to your reply! Thank you!

SlamSmallFish, 2023-07-06:
I meet the same problem too. Can anyone solve this problem？

Reverie10, 2023-07-14:
imuAccNoise: 0
  imuGyrNoise: 0
  imuAccBiasN: 0
  imuGyrBiasN: 0
If you set the parameters in your params.yaml file the same as the example given above, it will cause nan values to appear in the optimizer of the imupreintegration.cpp file, leading to optimization failure

Madhav2133, 2023-07-14:
Hi, I am facing a similar issue:

[image]

Can you please tell me if you were able to resolve this?

Dcasadoherraez, 2024-07-15:
Same here. Apparently it comes from NaN values computed by the imu preintegration:
```
[ WARN] [1721058402.293461363]: MSG to TF: Quaternion Not Properly Normalized
Error:   TF_NAN_INPUT: Ignoring transform for child_frame_id "base_link" from authority "unknown_publisher" because of a nan value in the transform (-nan -nan -nan) (-nan -nan -nan -nan)
         at line 240 in /tmp/binarydeb/ros-noetic-tf2-0.7.7/src/buffer_core.cpp
Error:   TF_DENORMALIZED_QUATERNION: Ignoring transform for child_frame_id "base_link" from authority "unknown_publisher" because of an invalid quaternion in the transform (-nan -nan -nan -nan)
         at line 255 in /tmp/binarydeb/ros-noetic-tf2-0.7.7/src/buffer_core.cpp

```