TixiaoShan/LIO-SAM #430: Do not understand the difference between the extrinsicRot and extrinsicRPY matrices in the params.yaml file

brandon-lawrence96, 2023-05-29:
Hi Trixiao! Thank you for sharing your amazing work! Can I ask what the difference between the extrinisicRot and extrinsicRPY is? Currently, to get from my imu frame to my lidar frame, I need to rotate my imu by -180 degrees about the x-axis. Does this mean that my extrinsicRot matrix should be 
[1, 0, 0,
  0, -1, 0,
  0, 0, -1]?

And how about the extrinsicRPY matrix? What should I set for that? 

Thank you for your assistance in this! :)

VineetTambe, 2023-08-22:
I had the same question! Does anyone know why?

grischi, 2023-08-23:
If you search for explanations you will find plenty of them in the issues if you also include closed ones in your search. Briefly summarized, they both rotate imu measurements from the corresponding measurement frame to the lidar frame (see imuConverter method in utility.hpp). The reason why there are two such rotations is that many imus output orientations (extrinsicRPY) in a different frame than angular velocities and accelerations (extrinsicRot).

XiaomingQu, 2023-10-11:
extrinisicRot rotates the acceleration and angular velocity of imu, extrinsicRPY rotates the geomagnetic direction of imu. These two are not the same coordinate system in the author's device.