TixiaoShan/LIO-SAM #433: Reconstruct jump and error

j953302, 2023-06-12:
Hello, I'm new to try LIO-SAM by my own data. Initially my Lidar TF drift a little but reconstruct doesn't failed (Really god bless). But after my AMR go straight it will jump and then fail reconstruct. Below shared the picture of my rviz and bag data. 

### Here is the some my data describe:
- Lidar : Velodyne VLP-16 Puck
- IMU : YAHboom 10 axis IMU
- Location: Indoor corridor
- ROS2 foxy
- data from AMR go straight and back straight

### the problem picture :
[image]

### the params.yaml:

> 

  ros__parameters:

    # Topics
    pointCloudTopic: "/points2"                   # Point cloud data
    imuTopic: "/imu"                        # IMU data
    odomTopic: "odometry/imu"                    # IMU pre-preintegration odometry, same frequency as IMU
    gpsTopic: "odometry/gpsz"                    # GPS odometry topic from navsat, see module_navsat.launch file

    # Frames
    lidarFrame: "velodyne"                                 #default : "lidar_link"
    baselinkFrame: "base_link"
    odometryFrame: "odom"
    mapFrame: "map"

    # GPS Settings
    useImuHeadingInitialization: false           # if using GPS data, set to "true"
    useGpsElevation: false                       # if GPS elevation is bad, set to "false"
    gpsCovThreshold: 2.0                         # m^2, threshold for using GPS data
    poseCovThreshold: 25.0                       # m^2, threshold for using GPS data

    # Export settings
    savePCD: false                               # https://github.com/TixiaoShan/LIO-SAM/issues/3
    savePCDDirectory: "/home/droxo/LIO-SAM-ros2/LOAM/"         # in your home folder, starts and ends with "/". Warning: the code deletes "LOAM" folder then recreates it. See "mapOptimization" for implementation

    # Sensor Settings
    sensor: velodyne                               # lidar sensor type, either 'velodyne', 'ouster' or 'livox'
    N_SCAN: 16                                   # number of lidar channels (i.e., Velodyne/Ouster: 16, 32, 64, 128, Livox Horizon: 6)
    Horizon_SCAN: 1800                            # lidar horizontal resolution (Velodyne:1800, Ouster:512,1024,2048, Livox Horizon: 4000)
    downsampleRate: 1                            # default: 1. Downsample your data if too many
    # points. i.e., 16 = 64 / 4, 16 = 16 / 1
    lidarMinRange: 1.0                           # default: 1.0, minimum lidar range to be used
    lidarMaxRange: 1000.0                        # default: 1000.0, maximum lidar range to be used

    # IMU Settings
    imuAccNoise: 3.9939570888238808e-03
    imuGyrNoise: 1.5636343949698187e-03
    imuAccBiasN: 6.4356659353532566e-05
    imuGyrBiasN: 3.5640318696367613e-05

    imuGravity: 9.80511
    imuRPYWeight: 0.01

    extrinsicTrans:  [ 0.0,  0.0, 0.0 ]
    extrinsicRot:    [1.0,  0.0,  0.0,
                       0.0,  1.0,  0.0,
                       0.0,  0.0,1.0 ]

    extrinsicRPY: [1.0,  0.0,  0.0,
                       0.0,  -0.259,  0.966,
                       0.0,  -0.966, -0.259 ] #degree -105 
    # LOAM feature threshold 
    edgeThreshold: 1.0
    surfThreshold: 0.1
    edgeFeatureMinValidNum: 10
    surfFeatureMinValidNum: 100

    # voxel filter paprams
    odometrySurfLeafSize: 0.2                     # default: 0.4 - outdoor, 0.2 - indoor
    mappingCornerLeafSize: 0.1                    # default: 0.2 - outdoor, 0.1 - indoor
    mappingSurfLeafSize: 0.2                     # default: 0.4 - outdoor, 0.2 - indoor

    # robot motion constraint (in case you are using a 2D robot)
    z_tollerance: 1000.0                          # meters
    rotation_tollerance: 1000.0                   # radians

    # CPU Params
    numberOfCores: 4                              # number of cores for mapping optimization
    mappingProcessInterval: 0.15                  # seconds, regulate mapping frequency

    # Surrounding map
    surroundingkeyframeAddingDistThreshold: 1.0   # meters, regulate keyframe adding threshold
    surroundingkeyframeAddingAngleThreshold: 0.2  # radians, regulate keyframe adding threshold
    surroundingKeyframeDensity: 2.0               # meters, downsample surrounding keyframe poses   
    surroundingKeyframeSearchRadius: 50.0         # meters, within n meters scan-to-map optimization
    # (when loop closure disabled)

    # Loop closure
    loopClosureEnableFlag: true
    loopClosureFrequency: 1.0                     # Hz, regulate loop closure constraint add frequency
    surroundingKeyframeSize: 50                   # submap size (when loop closure enabled)
    historyKeyframeSearchRadius: 15.0             # meters, key frame that is within n meters from
    # current pose will be considerd for loop closure
    historyKeyframeSearchTimeDiff: 30.0           # seconds, key frame that is n seconds older will be
    # considered for loop closure
    historyKeyframeSearchNum: 25                  # number of hostory key frames will be fused into a
    # submap for loop closure
    historyKeyframeFitnessScore: 0.3              # icp threshold, the smaller the better alignment

    # Visualization
    globalMapVisualizationSearchRadius: 1000.0    # meters, global map visualization radius
    globalMapVisualizationPoseDensity: 10.0       # meters, global map visualization keyframe density
    globalMapVisualizationLeafSize: 1.0           # meters, global map visualization cloud density

### ROS2 bag data include lidar data % imu data
ROS2 bag : https://drive.google.com/file/d/1mcWyffvXN4X3OiJ3ntP_UOecvU6WCeod/view?usp=sharing
lidar data : /points
imu data  : /imu

I have try change extrinsicRot and extrinsicRPY, and I have get a perfact model once. But only once. It was very tricky. Could anyone give me advise and if I have new finding update here. Thanks.

LHZ-code, 2023-06-20:
Hello, can I communicate with you about running lio-sam with ROS2? How can I get in touch with you?

j953302, 2023-06-20:
> Hello, can I communicate with you about running lio-sam with ROS2? How can I get in touch with you?

Sure. Could you give me your email? I could send you email in private.

LHZ-code, 2023-06-20:
> 

1337551819@qq.com

lukasj-imar, 2023-06-26:
Same problem here under ros2-humble. Do you have found the reason for that behaviour?

j953302, 2023-06-27:
@lukasj-imar Unfortunately no. But I change to use other person's open source which based on LIO-SAM.
I think the problem comes from IMU. But I didn't know the reason.

LHZ-code, 2023-06-27:
> @lukasj-imar Unfortunately no. But I change to use other person's open source which based on LIO-SAM. I think the problem comes from IMU. But I didn't know the reason.

可以解决这个问题嘛，那个开源的基于lio-sam的，如果可以的话，可以分享一下嘛

j953302, 2023-06-27:
@lukasj-imar  Sure. You could check li-slam-ros2
[https://github.com/rsasaki0109/li_slam_ros2](url)

lukasj-imar, 2023-06-29:
I found a solution for my problem, check: https://github.com/TixiaoShan/LIO-SAM/issues/437