TixiaoShan/LIO-SAM #437: Latency problem with ROS2 algorithm

lukasj-imar, 2023-06-29:
Hi everybody,

while testing ROS2-LIO-SAM on different machines, I encountered a problem related to the native high latency of the OS (Ubuntu 22.04 TLS, kernel: 5.19.0-45-generic).

As a result of the latency problem, LIO-SAM worked on one machine only with decreased replay rate of my own rosbags. On an other system (with much more powerfull hardware),  I had no chance to run LIO-SAM at any speed. It seems that point clouds won't be processed fast enough, so no LiDAR-odometry solution can be calculated with the consequence that ```imuPreintegration``` drifts away after a few seconds.

I fixed this problem by reducing the latency of the whole system. Therefor, I changed der generic kernel to a [lowlatency-kernel](https://wiki.ubuntuusers.de/Kernel/) and disabled switching-ability to high latency c-states by setting a [kernel boot parameter](https://www.kernel.org/doc/html/v5.0/admin-guide/pm/cpuidle.html) to ``ìdle=halt```.

Now, ROS2-LIO-SAM seems to work pretty well and stable on all of my machines. If you have issues with respect to performance, check out this fix and give me a feedback.

chivas1000, 2023-11-13:
Hi I might be ran into same issue as you.
My LIOSAM on ROS1 runs fine, but on the same setup, it runs like: https://github.com/TixiaoShan/LIO-SAM/issues/346
Also, I've debugged the code and I've seen although **ros2 topic delay** of the pointcloud topic are at around 10hz.
but in the Lidarpointcloud handler(the pointcloud callback function), it runs will around 7-9hz, sometimes 6hz.

Would you mind tell me how to switch generic kernel to a [lowlatency-kernel](https://wiki.ubuntuusers.de/Kernel/) and disabled switching-ability to high latency c-states by setting a [kernel boot parameter](https://www.kernel.org/doc/html/v5.0/admin-guide/pm/cpuidle.html) to ``ìdle=halt```? Or any I could refer?

Much appreciated!

lukasj-imar, 2023-12-05:
Hi @chivas1000,

sorry for my late response. Here are the steps you need to follow:

1. In ***/etc/default/grub*** edit the following parmeter to: ```GRUB_CMDLINE_LINUX_DEFAULT="quiet splash acpi=ht idle=halt intel_idle.max_cstate=0"```
2.  To start the ROS-System with RT-PRIO, edit in ***/etc/security/limits.conf*** the last entries, according to your user name 
<pre>#&lt;domain&gt;      &lt;type&gt;  &lt;item&gt;         &lt;value&gt;
#

#*               soft    core            0
#root            hard    core            100000
#*               hard    rss             10000
lukas		 -       rtprio		 98
lukas		 -	 nice		 -19
lukas		 -       memlock	 -1
</pre>

3. ```sudo update-grub```