TixiaoShan/LIO-SAM #441: IMU jittering when running with Ouster 64 LiDAR on ROS2 even after ptp time synchronization

klausmikealsontheking, 2023-07-12:
Hello TixiaoShan, I am trying to run LIOSAM using an Ouster 64 LiDAR and Xsens MTi 670G 9-axis IMU on ROS2 Foxy.  At first when the data was not visible on the rviz screen, I got to know that there was a time synchronization issue between my ouster and imu. After using the `sudo ptpd -i enxa0cec8fb57c3 -M -V` which synchronized the header stamps of both the /ouster/points and /imu/data, I could observe the LIOSAM data on the rviz screen. But then I started getting the below error.
`[lio_sam_imuPreintegration-3] [ERROR] [1689143066.422097210] [lio_sam_imuPreintegration]: Lookup would require extrapolation at time 1689143066.255501, but only time 1689143066.262720 is in the buffer, when looking up transform from frame [base_link] to frame [lidar_link]
[lio_sam_imuPreintegration-3] [WARN] [1689143067.443115554] [lio_sam_imuPreintegration]: Large velocity, reset IMU-preintegration!
`

https://github.com/TixiaoShan/LIO-SAM/assets/87766686/3e97b3d6-385d-4639-a55c-ed92c975a182

Please help me understand what the problem here is and help me rectify the issue. Given below is my param file.
/**:
  ros__parameters:

    # Topics
    pointCloudTopic: "/ouster/points"                   # Point cloud data
    imuTopic: "/imu/data"                        # IMU data
    odomTopic: "odometry/imu"                    # IMU pre-preintegration odometry, same frequency as IMU
    gpsTopic: "/gps_xsens"                    # GPS odometry topic from navsat, see module_navsat.launch file

    # Frames
    lidarFrame: "lidar_link"
    baselinkFrame: "base_link"
    odometryFrame: "odom"
    mapFrame: "map"

    # GPS Settings
    useImuHeadingInitialization: true          # if using GPS data, set to "true"
    useGpsElevation: false                       # if GPS elevation is bad, set to "false"
    gpsCovThreshold: 2.0                         # m^2, threshold for using GPS data
    poseCovThreshold: 25.0                       # m^2, threshold for using GPS data

    # Export settings
    savePCD: false                               # https://github.com/TixiaoShan/LIO-SAM/issues/3
    savePCDDirectory: "/Downloads/LOAM/"         # in your home folder, starts and ends with "/". Warning: the code deletes "LOAM" folder then recreates it. See "mapOptimization" for implementation

    # Sensor Settings
    sensor: ouster                               # lidar sensor type, either 'velodyne', 'ouster' or 'livox'
    N_SCAN: 64                                   # number of lidar channels (i.e., Velodyne/Ouster: 16, 32, 64, 128, Livox Horizon: 6)
    Horizon_SCAN: 1024                            # lidar horizontal resolution (Velodyne:1800, Ouster:512,1024,2048, Livox Horizon: 4000)
    downsampleRate: 1                            # default: 1. Downsample your data if too many
    # points. i.e., 16 = 64 / 4, 16 = 16 / 1
    lidarMinRange: 1.0                           # default: 1.0, minimum lidar range to be used
    lidarMaxRange: 1000.0                        # default: 1000.0, maximum lidar range to be used

    # IMU Settings
    imuAccNoise: 3.9939570888238808e-03
    imuGyrNoise: 1.5636343949698187e-03
    imuAccBiasN: 6.4356659353532566e-05
    imuGyrBiasN: 3.5640318696367613e-05

    imuGravity: 9.80511
    imuRPYWeight: 0.01

    extrinsicTrans:  [ 0.0,  0.0,  0.0 ]
    extrinsicRot:    [-1.0,  0.0,  0.0,
                       0.0,  1.0,  0.0,
                       0.0,  0.0, -1.0 ]
    extrinsicRPY: [ 0.0,  1.0,  0.0,
                   -1.0,  0.0,  0.0,
                    0.0,  0.0,  1.0 ]

    # LOAM feature threshold
    edgeThreshold: 1.0
    surfThreshold: 0.1
    edgeFeatureMinValidNum: 10
    surfFeatureMinValidNum: 100

    # voxel filter paprams
    odometrySurfLeafSize: 0.4                     # default: 0.4 - outdoor, 0.2 - indoor
    mappingCornerLeafSize: 0.2                    # default: 0.2 - outdoor, 0.1 - indoor
    mappingSurfLeafSize: 0.4                      # default: 0.4 - outdoor, 0.2 - indoor

    # robot motion constraint (in case you are using a 2D robot)
    z_tollerance: 1000.0                          # meters
    rotation_tollerance: 1000.0                   # radians

    # CPU Params
    numberOfCores: 4                              # number of cores for mapping optimization
    mappingProcessInterval: 0.15                  # seconds, regulate mapping frequency

    # Surrounding map
    surroundingkeyframeAddingDistThreshold: 1.0   # meters, regulate keyframe adding threshold
    surroundingkeyframeAddingAngleThreshold: 0.2  # radians, regulate keyframe adding threshold
    surroundingKeyframeDensity: 2.0               # meters, downsample surrounding keyframe poses   
    surroundingKeyframeSearchRadius: 50.0         # meters, within n meters scan-to-map optimization
    # (when loop closure disabled)

    # Loop closure
    loopClosureEnableFlag: true
    loopClosureFrequency: 1.0                     # Hz, regulate loop closure constraint add frequency
    surroundingKeyframeSize: 50                   # submap size (when loop closure enabled)
    historyKeyframeSearchRadius: 15.0             # meters, key frame that is within n meters from
    # current pose will be considerd for loop closure
    historyKeyframeSearchTimeDiff: 30.0           # seconds, key frame that is n seconds older will be
    # considered for loop closure
    historyKeyframeSearchNum: 25                  # number of hostory key frames will be fused into a
    # submap for loop closure
    historyKeyframeFitnessScore: 0.3              # icp threshold, the smaller the better alignment

    # Visualization
    globalMapVisualizationSearchRadius: 1000.0    # meters, global map visualization radius
    globalMapVisualizationPoseDensity: 10.0       # meters, global map visualization keyframe density
    globalMapVisualizationLeafSize: 1.0           # meters, global map visualization cloud density

klausmikealsontheking, 2023-07-13:
The problem was my extrensicRPY and Rot matrices. the problem got solved after I changed them to unit matrices

closed 2023-07-13 by klausmikealsontheking

ashBabu, 2024-03-20:
@klausmikealsontheking : will it be okay to share your ros2 bag?