TixiaoShan/LIO-SAM #450: how to get the full lidar frame pose but not only the key frame pose

Tartisan, 2023-09-19:
Thanks for the great works! I want the optimized lidar frame pose, but not only the key frame pose, what can I do?

rayee-github, 2023-09-19:
How about to increase keyframe frequency or decrease voxel filter rate?
Can you describe it in more detail.

XiaomingQu, 2023-10-11:
Changing [here](https://github.com/TixiaoShan/LIO-SAM/blob/master/src/mapOptmization.cpp#L251) to `if(true)` may output the feature points of each frame，but note that lio-sam is a real-time slam. You may need to ensure that the processor performance can keep up with the playback speed of the rosbag.

Robin6416KW, 2023-10-12:
hi @XiaomingQu,I testing the same sample file "walking_dataset.bag" with Lio-sam and Liorf.

the parameter in Lio-sam is original param.yaml file and Liorf is lio_sam_default.yaml. 
these two parameters is mosltly the same. 

In Lio-sam it output full lidar frame, but in Liorf it just output keyframe. 
I compare thoese two line you mention in Lio-sam and Liorf but found no difference.
Is there other reason make liorf just output keyframe? 
many thanks.

Tartisan, 2023-10-12:
> Changing [here](https://github.com/TixiaoShan/LIO-SAM/blob/master/src/mapOptmization.cpp#L251) and [here](https://github.com/TixiaoShan/LIO-SAM/blob/master/src/mapOptmization.cpp#L1354) to true may output the feature points of each frame，but note that lio-sam is a real-time slam. You may need to ensure that the processor performance can keep up with the playback speed of the rosbag.

Thanks, I konw.

closed 2023-10-12 by Tartisan