TixiaoShan/LIO-SAM #455: LIO-SAM [lio_sam_imuPreintegration-2] process has died [pid 32792, exit code -6

MingXuan-Xiong, 2023-10-13:
Hey,buddies,I encountered this issue while using LIO-SAM。Firstly，I run the launch file，then play the bag files which are provided by official website(like west.bag/casual_walk.bag) and reported an error ：
    **[lio_sam_imuPreintegration-2] process has died [pid 34480, exit code -6, cmd /home/xmx/catkin_ws/devel/lib/lio_sam/lio_sam_imuPreintegration __name:=lio_sam_imuPreintegration __log:=/home/xmx/.ros/log/27c241f0-6976-11ee-a9fd-89b191b964e1/lio_sam_imuPreintegration-2.log].**
log file: /home/xmx/.ros/log/27c241f0-6976-11ee-a9fd-89b191b964e1/lio_sam_imuPreintegration-2*.log
[lio_sam_imuPreintegration-2] restarting process
process[lio_sam_imuPreintegration-2]: started with pid [34642]
[ INFO] [1697166683.454453412]: ----> IMU Preintegration Started.
terminate called after throwing an instance of 'gtsam::IndeterminantLinearSystemException'
  what():  
Indeterminant linear system detected while working near variable
7061644215716937728 (Symbol: b0).

**Thrown when a linear system is ill-posed.  The most common cause for this
error is having underconstrained variables.  Mathematically, the system is
underdetermined.  See the GTSAM Doxygen documentation at
http://borg.cc.gatech.edu/ on gtsam::IndeterminantLinearSyst
emException for more information.**
    Any comments will be much appreciated。

Alexjgxt18, 2024-02-06:
Hi, did you solved it? I'm also trying with a bag file provided by the official site, specific the Livox one. I get the error only when I change the parameters of the lidar.

MingXuan-Xiong, 2024-03-21:
> hello,guys. how do you solve this problem?thanks
It's a gtsam version issue ，requested gtsam's version>=4.02 and i download the latest one

Saiteja-RoshAi, 2024-07-16:
> > hello,guys. how do you solve this problem?thanks
> > It's a gtsam version issue ，requested gtsam's version>=4.02 and i download the latest on
my version is 4.03 and still facing the error

Saiteja-RoshAi, 2024-07-16:
> I'm sitting on GTSAM 4.0.2 and I have exactly the same issue.
> 
> The data is coming from IsaacSim. Ouster lidar with a default 9DoF IMU. I had to solve the ring problem first in the imageProjection.cpp (i'm calculating the ring number as proposed). IMU calibration was not very complicated, as the IMU was rotated by 90 degrees in Y-axis by default and fixing that fixed the orientation.
> 
> LIO-SAM is publishing in the map topic from time to time. I can see the pointclouds in the rviz, global map and local map are appearing for a while from time to time, but they are not complete - they are more like 1/4 of the map which is strange.
> 
> I thought that it could be the GTSAM issue, so I compiled the 4.0.2 version, recompiled the LIO-SAM, but still I've got the same problem. Process has died.
> 
> ```
> [ WARN] [1713281915.576376555, 5.129999885]: TF_REPEATED_DATA ignoring data with redundant timestamp for frame lidar_link (parent odom) at time 1294.749972 according to authority unknown_publisher
> 
> 278         at line  in /tmp/binarydeb/ros-noetic-tf2-0.7.7/src/buffer_core.cpp278
>  in /tmp/binarydeb/ros-noetic-tf2-0.7.7/src/buffer_core.cpp
> Warning: TF_REPEATED_DATA ignoring data with redundant timestamp for frame lidar_link (parent odom) at time 1294.749972 according to authority unknown_publisher
>          at line 278 in /tmp/binarydeb/ros-noetic-tf2-0.7.7/src/buffer_core.cpp
>   what():  
> Indeterminant linear system detected while working near variable
> 8646911284551352320 (Symbol: x0).
> 
> Thrown when a linear system is ill-posed.  The most common cause for this
> error is having underconstrained variables.  Mathematically, the system is
> underdetermined.  See the GTSAM Doxygen documentation at
> http://borg.cc.gatech.edu/ on gtsam::IndeterminantLinearSystemException for
> more information.
> [lio_sam_imuPreintegration-1] process has died [pid 23544, exit code -6, cmd /root/catkin_ws/devel/lib/lio_sam/lio_sam_imuPreintegration __name:=lio_sam_imuPreintegration __log:=/root/.ros/log/8b2bdfe4-fc01-11ee-a2b7-0242ac130002/lio_sam_imuPreintegration-1.log].
> log file: /root/.ros/log/8b2bdfe4-fc01-11ee-a2b7-0242ac130002/lio_sam_imuPreintegration-1*.log
> [lio_sam_imuPreintegration-1] restarting process
> process[lio_sam_imuPreintegration-1]: started with pid [23576]
> ```
How didi you solve the ring problem in imupreintegration?

wxyLuna2024, 2024-07-25:
> How didi you solve the ring problem in imupreintegration?

refer to #452