TixiaoShan/LIO-SAM #456: The map is constructed incorrectly.

LouiMilvus, 2023-10-18:
imu outputs 400 hertz, OusterOS64 lidar.
Configuration settings:
sensor: ouster                               # lidar sensor type, either 'velodyne', 'ouster' or 'livox'
N_SCAN: 64                                   # number of lidar channels (i.e., Velodyne/Ouster: 16, 32, 64, 128, Livox Horizon: 6)
Horizon_SCAN: 1024                            # lidar horizontal resolution (Velodyne:1800, Ouster:512,1024,2048, Livox Horizon: 4000)
downsampleRate: 1                            # default: 1. Downsample your data if too many
# points. i.e., 16 = 64 / 4, 16 = 16 / 1
lidarMinRange: 1.0                           # default: 1.0, minimum lidar range to be used
lidarMaxRange: 1000.0                        # default: 1000.0, maximum lidar range to be used

# IMU Settings
  # IMU Settings
imuAccNoise: 3.9939570888238808e-01
imuGyrNoise: 1.5636343949698187e-01
imuAccBiasN: 6.4356659353532566e-03
imuGyrBiasN: 3.5640318696367613e-03

imuGravity: 9.80511
imuRPYWeight: 0.01

extrinsicTrans:  [ 0.0,  0.0,  0.0 ]
extrinsicRot:    [ 0.0,  1.0,  0.0,
               -1.0,  0.0,  0.0,
                0.0,  0.0,  1.0 ]
extrinsicRPY: [ 0.0,  1.0,  0.0,
               -1.0,  0.0,  0.0,
                0.0,  0.0,  1.0 ]
Part of the terminal output:
[lio_sam_imuPreintegration-3] [ERROR] [1697604124.237840049] [lio_sam_imuPreintegration]: Lookup would require extrapolation into the past.  Requested time 1679940531.898065 but the earliest data is at ti
me 1679940531.953213, when looking up transform from frame [base_link] to frame [lio_laser_data_frame]                                                                                                      
[lio_sam_imuPreintegration-3] [ERROR] [1697604124.242714477] [lio_sam_imuPreintegration]: Lookup would require extrapolation into the past.  Requested time 1679940531.898065 but the earliest data is at ti
me 1679940531.953213, when looking up transform from frame [base_link] to frame [lio_laser_data_frame]                                                                                                      
[lio_sam_imuPreintegration-3] [WARN] [1697604125.049085340] [lio_sam_imuPreintegration]: Large velocity, reset IMU-preintegration!                                                                          
[lio_sam_imuPreintegration-3] [WARN] [1697604125.863422819] [lio_sam_imuPreintegration]: Large velocity, reset IMU-preintegration!                                                                          
[lio_sam_imuPreintegration-3] [WARN] [1697604126.236174261] [lio_sam_imuPreintegration]: Large velocity, reset IMU-preintegration!                                                                          
[lio_sam_imuPreintegration-3] [WARN] [1697604127.064230519] [lio_sam_imuPreintegration]: Large velocity, reset IMU-preintegration!                                                                          
[lio_sam_imuPreintegration-3] [WARN] [1697604127.860578113] [lio_sam_imuPreintegration]: Large velocity, reset IMU-preintegration!                                                                          
[lio_sam_imuPreintegration-3] [WARN] [1697604128.245528674] [lio_sam_imuPreintegration]: Large velocity, reset IMU-preintegration!                                                                          
[lio_sam_imuPreintegration-3] [WARN] [1697604129.068931595] [lio_sam_imuPreintegration]: Large velocity, reset IMU-preintegration!                                                                          
[lio_sam_imuPreintegration-3] [WARN] [1697604129.888684117] [lio_sam_imuPreintegration]: Large velocity, reset IMU-preintegration!                                                                          
[lio_sam_imuPreintegration-3] [WARN] [1697604130.662336353] [lio_sam_imuPreintegration]: Large velocity, reset IMU-preintegration!                                                                          
[lio_sam_imuPreintegration-3] [WARN] [1697604131.447464140] [lio_sam_imuPreintegration]: Large velocity, reset IMU-preintegration!                                                                          
[lio_sam_imuPreintegration-3] [WARN] [1697604132.248969047] [lio_sam_imuPreintegration]: Large velocity, reset IMU-preintegration!                                                                          
[lio_sam_imuPreintegration-3] [WARN] [1697604132.646798983] [lio_sam_imuPreintegration]: Large velocity, reset IMU-preintegration!                                                                          
[lio_sam_imuPreintegration-3] [WARN] [1697604133.455351094] [lio_sam_imuPreintegration]: Large velocity, reset IMU-preintegration!                                                                          
[rviz2-7] [INFO] [1697604133.664406161] [rviz2]: Message Filter dropping message: frame 'lio_laser_data_frame' at time 1679940531.298 for reason 'the timestamp on the message is earlier than all the data 
in the transform cache'                                                                                                                                                                                     
[rviz2-7] [INFO] [1697604133.664793542] [rviz2]: Message Filter dropping message: frame 'lio_laser_data_frame' at time 1679940531.398 for reason 'the timestamp on the message is earlier than all the data 
in the transform cache'                                                                                                                                                                                     
[lio_sam_imuPreintegration-3] [WARN] [1697604133.848509198] [lio_sam_imuPreintegration]: Large velocity, reset IMU-preintegration!                                                                          
[lio_sam_imuPreintegration-3] [WARN] [1697604134.687825062] [lio_sam_imuPreintegration]: Large velocity, reset IMU-preintegration!

Alexjgxt18, 2024-02-07:
Hi, did you solve it?

LouiMilvus, 2024-03-15:
> Hi, did you solve it?

Good day! No, this issue still persists. I recorded another ROS2 bag, the IMU sensor is positioned differently (I changed the matrix), yet the map still jumps.