TixiaoShan/LIO-SAM #458: processing 2 time stamps old point cloud

huzefablue, 2023-10-24:
In imageProjection.cpp in the cachePointCloud() function, at the very start, the cloudQueue is not processed till it has 2 elements. And after that is processes the oldest element in time in the queue. Wouldn't this cause the odometry to always be outdated by two timestamps?

XiaomingQu, 2023-11-26:
If the pose you accept comes from "lio_sam/odomTopic" in the yaml format configuration file, then its timestamp is almost current, because the function "IMUPreintegration::imuHandler" integrates it in real time and sends it out with this topic. 
Of course, the imu's bias is two timestamps behind, but it doesn't matter. 
I think setting the cloudQueue cache size to 0 will not affect the operation.

Sonne-Zhu, 2024-05-09:
Set the cloudQueue cache size to 0 will lead bad result, why？