TixiaoShan/LIO-SAM #468: ros2 galactic unknown Cmake command "rosidl_get_typesupport_target"

123Bruceche, 2023-12-28:
when I colcon build at galactic 20.04 ,I got this mistake.  please help

cmake Error Cmakelists.txt 43 (rosidl_get_typesupport_target):
Unknown Cmake command rosidl_get_typesupport_target

cyhunblr, 2024-01-21:
Did you solve?

godic97, 2024-01-26:
Hi!

I solved this issue. (foxy)

First, there's no rosidl_get_typesupport_target function in foxy.
So, visit to [here](https://github.com/ros2/rosidl/blob/rolling/rosidl_cmake/cmake/rosidl_get_typesupport_target.cmake), and replace the function to the raw code.

Second, in my case, tf2_eigen and tf2_geometry_msgs are .h, not .hpp.
So, rewrite lib name in src/include/lio_sam/utility.hpp.

Alexjgxt18, 2024-02-08:
Hi @godic97 , how do you 
> replace the function to the raw code in CMakeLists.txt .

 I just pasted the code from the link, after this line `rosidl_generate_interfaces(${PROJECT_NAME} "msg/CloudInfo.msg" "srv/SaveMap.srv" DEPENDENCIES std_msgs sensor_msgs) ` and got the files name error. Once I made colcon build I got fatal error: lio_sam/msg/cloud_info.hpp: No such file or directory

    2 | #include "lio_sam/msg/cloud_info.hpp"
      |          ^~~~~~~~~~~~~~~~~~~~~~~~~~~~
So... I think I made something wrong

InguChoi, 2024-04-03:
Hi, I solve the issue with reference to @Alexjgxt18.
(https://github.com/ros2/rosidl/blob/rolling/rosidl_cmake/cmake/rosidl_get_typesupport_target.cmake)

[image]

[image]

tang-nhan, 2024-04-24:
> Hi @godic97 , how do you
> 
> > replace the function to the raw code in CMakeLists.txt .
> 
> I just pasted the code from the link, after this line `rosidl_generate_interfaces(${PROJECT_NAME} "msg/CloudInfo.msg" "srv/SaveMap.srv" DEPENDENCIES std_msgs sensor_msgs) ` and got the files name error. Once I made colcon build I got
> 
> ```
> fatal error: lio_sam/msg/cloud_info.hpp: No such file or directory
> 
>     2 | #include "lio_sam/msg/cloud_info.hpp"
>       |          ^~~~~~~~~~~~~~~~~~~~~~~~~~~~
> ```
> 
> So... I think I did something wrong

Hi @Alexjgxt18, I also met this issue. Have you solved it?

dal88236, 2024-11-07:
Hi all, 
Following is the CMakeLists.txt that works for me. My ROS2 version is galactic.

```
cmake_minimum_required(VERSION 3.5)
project(lio_sam)

if(NOT CMAKE_BUILD_TYPE AND NOT CMAKE_CONFIGURATION_TYPES)
  set(CMAKE_BUILD_TYPE Release)
endif()

# find dependencies
find_package(ament_cmake REQUIRED)
find_package(rosidl_default_generators REQUIRED)
find_package(rclcpp REQUIRED)
find_package(rclpy REQUIRED)
find_package(cv_bridge REQUIRED)
find_package(std_msgs REQUIRED)
find_package(sensor_msgs REQUIRED)
find_package(geometry_msgs REQUIRED)
find_package(nav_msgs REQUIRED)
find_package(visualization_msgs REQUIRED)
find_package(pcl_conversions REQUIRED)
find_package(pcl_msgs REQUIRED)
find_package(tf2 REQUIRED)
find_package(tf2_geometry_msgs REQUIRED)
find_package(tf2_sensor_msgs REQUIRED)
find_package(tf2_eigen REQUIRED)
find_package(tf2_ros REQUIRED)
set(OpenCV_STATIC ON)
find_package(OpenCV REQUIRED)
find_package(PCL REQUIRED)
find_package(GTSAM REQUIRED)
find_package(Eigen REQUIRED)
find_package(OpenMP REQUIRED)

include_directories(
  include/lio_sam
)

rosidl_generate_interfaces(${PROJECT_NAME} "msg/CloudInfo.msg" "srv/SaveMap.srv" DEPENDENCIES std_msgs sensor_msgs)

add_executable(${PROJECT_NAME}_featureExtraction src/featureExtraction.cpp)
ament_target_dependencies(${PROJECT_NAME}_featureExtraction rclcpp rclpy std_msgs sensor_msgs geometry_msgs nav_msgs pcl_conversions pcl_msgs visualization_msgs tf2 tf2_ros tf2_eigen tf2_sensor_msgs tf2_geometry_msgs OpenCV PCL)
# rosidl_get_typesupport_target(cpp_typesupport_target ${PROJECT_NAME} "rosidl_typesupport_cpp")
rosidl_target_interfaces(${PROJECT_NAME}_featureExtraction ${PROJECT_NAME} "rosidl_typesupport_cpp")

add_executable(${PROJECT_NAME}_imageProjection src/imageProjection.cpp)
ament_target_dependencies(${PROJECT_NAME}_imageProjection rclcpp rclpy std_msgs sensor_msgs geometry_msgs nav_msgs pcl_conversions pcl_msgs visualization_msgs tf2 tf2_ros tf2_eigen tf2_sensor_msgs tf2_geometry_msgs OpenCV PCL)
rosidl_target_interfaces(${PROJECT_NAME}_imageProjection ${PROJECT_NAME} "rosidl_typesupport_cpp")

add_executable(${PROJECT_NAME}_imuPreintegration src/imuPreintegration.cpp)
ament_target_dependencies(${PROJECT_NAME}_imuPreintegration rclcpp rclpy std_msgs sensor_msgs geometry_msgs nav_msgs pcl_conversions pcl_msgs visualization_msgs tf2 tf2_ros tf2_eigen tf2_sensor_msgs tf2_geometry_msgs OpenCV PCL GTSAM Eigen)
target_link_libraries(${PROJECT_NAME}_imuPreintegration gtsam)
rosidl_target_interfaces(${PROJECT_NAME}_imuPreintegration ${PROJECT_NAME} "rosidl_typesupport_cpp")

add_executable(${PROJECT_NAME}_mapOptimization src/mapOptmization.cpp)
ament_target_dependencies(${PROJECT_NAME}_mapOptimization rclcpp rclpy std_msgs sensor_msgs geometry_msgs nav_msgs pcl_conversions pcl_msgs visualization_msgs tf2 tf2_ros tf2_eigen tf2_sensor_msgs tf2_geometry_msgs OpenCV PCL GTSAM)
if (OpenMP_CXX_FOUND)
  target_link_libraries(${PROJECT_NAME}_mapOptimization gtsam "${cpp_typesupport_target}" OpenMP::OpenMP_CXX)
  rosidl_target_interfaces(${PROJECT_NAME}_mapOptimization ${PROJECT_NAME} "rosidl_typesupport_cpp")
else()
  target_link_libraries(${PROJECT_NAME}_mapOptimization gtsam "${cpp_typesupport_target}")
  rosidl_target_interfaces(${PROJECT_NAME}_mapOptimization ${PROJECT_NAME} "rosidl_typesupport_cpp")
endif()

install(
  DIRECTORY launch
  DESTINATION share/${PROJECT_NAME}/
)

install(
  DIRECTORY config
  DESTINATION share/${PROJECT_NAME}/
)

install(
  TARGETS ${PROJECT_NAME}_imageProjection
  DESTINATION lib/${PROJECT_NAME}
)

install(
  TARGETS ${PROJECT_NAME}_imuPreintegration
  DESTINATION lib/${PROJECT_NAME}
)

install(
  TARGETS ${PROJECT_NAME}_featureExtraction
  DESTINATION lib/${PROJECT_NAME}
)

install(
  TARGETS ${PROJECT_NAME}_mapOptimization
  DESTINATION lib/${PROJECT_NAME}
)

install(
  DIRECTORY "include/"
  DESTINATION include
)

ament_export_include_directories(include)

if(BUILD_TESTING)
  find_package(ament_lint_auto REQUIRED)
  # the following line skips the linter which checks for copyrights
  # uncomment the line when a copyright and license is not present in all source files
  #set(ament_cmake_copyright_FOUND TRUE)
  # the following line skips cpplint (only works in a git repo)
  # uncomment the line when this package is not in a git repo
  #set(ament_cmake_cpplint_FOUND TRUE)
  ament_lint_auto_find_test_dependencies()
endif()

ament_package()
```

Also, as mentioned above, in utility.hpp, line 40 and 41 shall be changed to 
```
#include <tf2_eigen/tf2_eigen.h>
#include <tf2_geometry_msgs/tf2_geometry_msgs.h>
```

FW88888888, 2024-11-14:
Hi, I solve this issue with same to @InguChoi

CESYouth, 2025-03-30:
谢谢，我也在foxy版本，成功解决了这个问题

cangrong, 2025-04-24:
Thank you very much!

ccpdead, 2025-04-27:
Thank you very much
it's work ok...

Finished <<< lio_sam [10.6s]

Summary: 3 packages finished [15.5s]
  1 package had stderr output: lio_sam

EthanTong1, 2025-08-09:
> > 嗨[@godic97](https://github.com/godic97),你怎么
> > > 将函数替换为CMakeLists.txt中的原始代码。
> > 
> > 
> > 我刚刚从链接中粘贴了代码，在这一行之后`rosidl_generate_interfaces(${PROJECT_NAME} "msg/CloudInfo.msg" "srv/SaveMap.srv" DEPENDENCIES std_msgs sensor_msgs) `并得到了文件名错误。一旦我做了colcon建立我得到了
> > ```
> > fatal error: lio_sam/msg/cloud_info.hpp: No such file or directory
> > 
> >     2 | #include "lio_sam/msg/cloud_info.hpp"
> >       |          ^~~~~~~~~~~~~~~~~~~~~~~~~~~~
> > ```
> > 
> > 
> >     
> >       
> >     
> > 
> >       
> >     
> > 
> >     
> >   
> > 所以...我想我做错了什么
> 
> 嗨[@Alexjgxt18](https://github.com/Alexjgxt18),我也遇到了这个问题。你解决了吗？

> Hi [@Alexjgxt18](https://github.com/Alexjgxt18), I also met this issue. Have you solved it?

Hello, I have also encountered the same problem. How did you solve it?