TixiaoShan/LIO-SAM #469: Run LIO-SAM with Carla and ROS2

guzgonav, 2024-01-07:
Hi,
I've been trying to use LIO-SAM to map a ros2bag recorded in carla, however I'm not able to make it work. 
The topics I have are the following ones: 
```
/carla/ego_vehicle/control/set_transform
/carla/ego_vehicle/enable_autopilot
/carla/ego_vehicle/gnss
/carla/ego_vehicle/imu
/carla/ego_vehicle/lidar
/carla/ego_vehicle/odometry
/carla/ego_vehicle/rgb_front/camera_info
/carla/ego_vehicle/rgb_front/image
/carla/ego_vehicle/rgb_view/camera_info
/carla/ego_vehicle/rgb_view/image
/carla/ego_vehicle/speedometer
/carla/ego_vehicle/vehicle_control_manual_override
/carla/map
/carla/markers
/carla/markers/static
/clock
/events/read_split
/events/write_split
/parameter_events
/rosout
/tf
```
So I've changes the params.yaml topics settings to: 
`# Topics
    pointCloudTopic: "/points_raw"                   # Point cloud data
    imuTopic: "/imu_raw"                        # IMU data
    odomTopic: "odometry/imu"                    # IMU pre-preintegration odometry, same frequency as IMU
    gpsTopic: "odometry/gpsz"                    # GPS odometry topic from navsat, see module_navsat.launch file
`

With that changes, the SLAM algorithm doesn't work, what else do I need to modify? 

Thanks in advance :)

TZECHIN6, 2024-01-28:
Can u clarify how it doesn't work? Miss aligned or even no map is built

guzgonav, 2024-03-28:
It didn't show map at all, I finally made it work changing the following in config/params.yaml

```
    # Topics
    pointCloudTopic: "/carla/ego_vehicle/lidar"                   # Point cloud data
    imuTopic: "/carla/ego_vehicle/imu"                        # IMU data
    odomTopic: "/carla/ego_vehicle/odometry"                    # IMU pre-preintegration odometry, same frequency as IMU
    gpsTopic: "/carla/ego_vehicle/gnss"                    # GPS odometry topic from navsat, see module_navsat.launch file

    # Frames
    lidarFrame: "ego_vehicle/lidar"
    baselinkFrame: "ego_vehicle"
    odometryFrame: "odom"
    mapFrame: "map"
```

Tharun7102, 2025-02-25:
> It didn't show map at all, I finally made it work changing the following in config/params.yaml
>
> ```
>     # Topics
>     pointCloudTopic: "/carla/ego_vehicle/lidar"                   # Point cloud [...]

Other than this, did you make any other changes ? 
I am trying with ROS noetic, but unable to build map ..!

ajay1606, 2025-04-03:
@Tharun7102 Can you share exact error message, will be helpful to understand actual problem.