TixiaoShan/LIO-SAM #474: Standing ego vehicle moves in rviz

weisenberg, 2024-02-05:
Hi. I run the algorithm on ROS2 with my recorded ros bag file from Carla (Velodyne LiDAR.) I have the IMU noise values from the imu_utils package and it seems right. However, to test it on the lio_sam algorithm, I record a stationary car and replay the bag, but the car (after a while) starts to drift and fly around rapidly with time. I can solve this by manually setting this value in imuPreintegration.cpp:
```c++
gtsam::NavState prevStateOdom(gtsam::Pose3(), gtsam::Velocity3(0.0, 0.0, 0.0));
``` 
This prevents the transformation for lidar_link to be anything other than the specified values above. However, to find the root of this issue I get nowhere. Does anyone have any idea?

LouiMilvus, 2024-03-15:
Good day! Where exactly did you make the changes?

weisenberg, 2024-04-02:
Hi! you mean changes to run Carla on lio_sam or changes to solve this issue? The latter got solved by setting the world tick to a higher frequency.