TixiaoShan/LIO-SAM #479: ERROR: cannot launch node of type [lio_sam/lio_sam_imuPreintegration]: Cannot locate node of type [lio_sam_imuPreintegration] in package [lio_sam]. Make sure file exists in package path and permission is set to executable (chmod +x)

Sisyphus-neo, 2024-03-16:
ERROR: cannot launch node of type [lio_sam/lio_sam_imuPreintegration]: Cannot locate node of type [lio_sam_imuPreintegration] in package [lio_sam]. Make sure file exists in package path and permission is set to executable (chmod +x)
ERROR: cannot launch node of type [lio_sam/lio_sam_imageProjection]: Cannot locate node of type [lio_sam_imageProjection] in package [lio_sam]. Make sure file exists in package path and permission is set to executable (chmod +x)
ERROR: cannot launch node of type [lio_sam/lio_sam_featureExtraction]: Cannot locate node of type [lio_sam_featureExtraction] in package [lio_sam]. Make sure file exists in package path and permission is set to executable (chmod +x)
ERROR: cannot launch node of type [lio_sam/lio_sam_mapOptmization]: Cannot locate node of type [lio_sam_mapOptmization] in package [lio_sam]. Make sure file exists in package path and permission is set to executable (chmod +x)

lwdone, 2024-03-23:
I also encountered this problem. Have you solved it? I started getting errors at 80% of catkin_make. Although it finally reached 100%, there were many compilation errors in the middle.

ahmedrh99, 2024-08-28:
Can you share how you managed to solve the problem if you did.