TixiaoShan/LIO-SAM #480: I don't find the topic sub for GPS

Spirquel, 2024-03-20:
Hello everyone,
I search in the code of this slam and i don't find the sub GPS like a "sensor_msgs/NavSatFix". 
Does anyone know what is the file where is this information ? Please.

XiaomingQu0913, 2024-04-13:
GPS may be filtered by navsat inside ros, and the smoothed GPS is located in the topic specified by gpsTopic in src/LIO-SAM/config/params.yaml

Spirquel, 2024-04-16:
oh okay, in others SLAM the msg for gnss is "NavSatFix" but here the msg gps is "nav_msgs/msg/Odometry" by a node transformation with the original msg "NavSatFix" to "nav_msgs/msg/Odometry". Thank, I understand the mistake I have, I didn't give much importance to the param file.

closed 2024-04-16 by Spirquel