TixiaoShan/LIO-SAM #481: Rooftop rosbag working in ROS2

ashBabu, 2024-03-20:
Hi, I am in ROS2 humble, ubuntu 22.04
I have downloaded the [rooftop_ouster_dataset.bag](https://drive.google.com/drive/folders/1gJHwfdHCRdjP7vuT556pv8atqrCJPbUq) from here. Then I used [rosbags](https://ternaris.gitlab.io/rosbags/topics/convert.html#usage) to convert ROS1 bag to ROS2 bag. My params file is as follows (only the ones I changed)
```
/**:
  ros__parameters:

    # Topics
    pointCloudTopic: "/points_raw"                   # Point cloud data
    imuTopic: "/imu_raw"                        # IMU data
    odomTopic: "odometry/imu"                    # IMU pre-preintegration odometry, same frequency as IMU
    gpsTopic: "odometry/gpsz"                    # GPS odometry topic from navsat, see module_navsat.launch file

    # Frames
    lidarFrame: "os_lidar"
    baselinkFrame: "base_link"
    odometryFrame: "odom"
    mapFrame: "map"

    # GPS Settings
    useImuHeadingInitialization: false           # if using GPS data, set to "true"
    useGpsElevation: false                       # if GPS elevation is bad, set to "false"
    gpsCovThreshold: 2.0                         # m^2, threshold for using GPS data
    poseCovThreshold: 25.0                       # m^2, threshold for using GPS data

    # Export settings
    savePCD: false                               # https://github.com/TixiaoShan/LIO-SAM/issues/3
    savePCDDirectory: "/Downloads/LOAM/"         # in your home folder, starts and ends with "/". Warning: the code deletes "LOAM" folder then recreates it. See "mapOptimization" for implementation

    # Sensor Settings
    sensor: ouster                               # lidar sensor type, either 'velodyne', 'ouster' or 'livox'
    N_SCAN: 32                                   # number of lidar channels (i.e., Velodyne/Ouster: 16, 32, 64, 128, Livox Horizon: 6)
    Horizon_SCAN: 512                            # lidar horizontal resolution (Velodyne:1800, Ouster:512,1024,2048, Livox Horizon: 4000)
    downsampleRate: 1                            # default: 1. Downsample your data if too many
    # points. i.e., 16 = 64 / 4, 16 = 16 / 1
    lidarMinRange: 1.0                           # default: 1.0, minimum lidar range to be used
    lidarMaxRange: 1000.0 
```

I have used a static transform publisher to link `lidar_link` to `os_lidar` as this is the frame id of the `/points_raw`. 

*  `ros2 bag play my_bagfile`
*  `ros2 launch lio_sam run.launch.py`

My tf tree
[image]

There's nothing shown in Rviz
[image]

```
ros2 topic echo /imu_raw --once
header:
  stamp:
    sec: 1594151157
    nanosec: 468152549
  frame_id: imu_link
orientation:
  x: 0.11496150493621826
  y: -0.151968315243721
  z: -0.712364673614502
  w: 0.6754450798034668
orientation_covariance:
- 0.01
- 0.0
- 0.0
- 0.0
- 0.01
- 0.0
- 0.0
- 0.0
- 0.01
angular_velocity:
  x: -0.29357796907424927
  y: 0.021105529740452766
  z: 0.48917731642723083
angular_velocity_covariance:
- 0.01
- 0.0
- 0.0
- 0.0
- 0.01
- 0.0
- 0.0
- 0.0
- 0.01
linear_acceleration:
  x: 2.368850171416998
  y: 0.7340992871671915
  z: -8.511016484498978
linear_acceleration_covariance:
- 0.01
- 0.0
- 0.0
- 0.0
- 0.01
- 0.0
- 0.0
- 0.0
- 0.01
```

```
 ros2 topic echo /points_raw --once
header:
  stamp:
    sec: 1594151229
    nanosec: 817840896
  frame_id: os_lidar
height: 128
width: 1024
fields:
- name: x
  offset: 0
  datatype: 7
  count: 1
- name: y
  offset: 4
  datatype: 7
  count: 1
- name: z
  offset: 8
  datatype: 7
  count: 1
- name: intensity
  offset: 16
  datatype: 7
  count: 1
- name: t
  offset: 20
  datatype: 6
  count: 1
- name: reflectivity
  offset: 24
  datatype: 4
  count: 1
- name: ring
  offset: 26
  datatype: 2
  count: 1
- name: noise
  offset: 28
  datatype: 4
  count: 1
- name: range
  offset: 32
  datatype: 6
  count: 1
is_bigendian: false
point_step: 48
row_step: 49152
data:
- 203
- 29
- 129
- 188
- 11
- 78
- 155
- 186
- 0
- 0
- 0
- 0
- 0
- 0
- 0
- 0
- 0
- 0
- 192
- 64
- 0
- 0
- 0
- 0
- 0
- 0
- 0
- 0
- 140
- 3
- 0
- 0
- 0
- 0
- 0
- 0
- 0
- 0
- 0
- 0
- 212
- 17
- 183
- 106
- 209
- 127
- 0
- 0
- 227
- 111
- 129
- 188
- 151
- 91
- 215
- 185
- 0
- 0
- 0
- 0
- 0
- 0
- 0
- 0
- 0
- 0
- 192
- 64
- 0
- 0
- 0
- 0
- 0
- 0
- 1
- 0
- 96
- 3
- 0
- 0
- 0
- 0
- 0
- 0
- 0
- 0
- 0
- 0
- 212
- 17
- 183
- 106
- 209
- 127
- 0
- 0
- 8
- 114
- 129
- 188
- 231
- 233
- 193
- 57
- 0
- 0
- 0
- 0
- 0
- 0
- 0
- 0
- 0
- 0
- 160
- 64
- 0
- 0
- 0
- 0
- 0
- 0
- 2
- 0
- 10
- 3
- 0
- 0
- '...'
is_dense: true
---
```

Any pointers to where am I going wrong? Thanks in advance

Spirquel, 2024-03-21:
the problem is your tf tree. In Rviz, for seen the map, it show the tf "map". But here your frame map isn't active/powered. for the active this map, you must to create a tf transform dynamic with the tf odom and the base_link. Like that, your tf map will be active. Because actually, the point of your robot isn't send at your tf map, so the map don't create cause she not receives points.

ashBabu, 2024-03-21:
@Spirquel : Thanks for the reply. However I am still a bit confused here. The `odom --> base_link` transform is to be published by the SLAM system, isn't? What I observed after running the LIO_SAM is that it is not being published for reasons unknown to me yet.
The `map-->odom` is a static transform published by the `run.launch.py` and hence it's available as a detached one in the TF tree

wxyLuna2024, 2024-07-22:
> @Spirquel : Thanks for the reply. However I am still a bit confused here. The `odom --> base_link` transform is to be published by the SLAM system, isn't? What I observed after running the LIO_SAM [...]

Hi, have you resolve this? I have exact same problem as you. My mapped trajectory is shown, but there's no concatenated points. I wonder if this problem happens related to ros2 bag
[image]

Spirquel, 2024-07-22:
For me the problem is the name of yours frame. Because if you have the same problem, in use "ros2 topic echo /[name of your topic]" you see this data. For know the name of your topic use by ros2bag, you have to use "ros2 bag info [name_bag].db3". But if you test to use "ros2 topic /echo [name_topic_lio_sam]" i think you see nothing, because your lio sam is not link with your bag. For see the connections with your topic and node, you can to use "rqt_graph" is a command for show these connection and for know if your topic bag and lio-sam are connected. For me the problem is here : 
[image]
[image]
these frame are used for get your tree frame. So you have to change the name in the param.yaml with your name frame, and change the name "child_frame_id" by your lidar_name frame or maybe your base_link. And for your "map trajectory" I assumed is your imu trajectory and no the slam trajectory isn't it ?

wxyLuna2024, 2024-07-22:
Hi, thank you so much for the speedy reply. Yes I didn't thought of maybe the trajectory is just a visualization of my imu trajectory. If that's the case that would mean I never really initiated the correct mapping process. As I did ros2 bag info my-bag.db3, these are the topics I got:
[image]
And accordingly, I have my params.yaml file as follows:
[image]
And the mapoptimization:
[image] 
I am trying to run kitti360 ros2 bag that I converted by myself. It only contains point cloud data, imu, and gps/velo, gps/fixed. As I did ros2 topic echo, I could see all four topics publishing stuff(not empty) help will be greatly appreciated!!

Spirquel, 2024-07-22:
Okay, so your bag isn't a problem. The problem is the connection with the slam. Can you use "rqt_graph" when your launch your bag and your slam (it's for show all connections in the ros system). And send with that your tf_tree (with "ros2 run tf2_tools view_frames" or "ros2 run rqt_tf-tree rqt_tf_tree (it's like rqt_graph but for frames)). Also, you have nothing msg by your slam ? like a problem with odom frame or with the time ?

wxyLuna2024, 2024-07-22:
Hello,
[image]
As I run ros2 run tf2_tools view_frames, I got error like this 
[image]
And nothing is showing up in my rviz2.
[image]
As I launch the launch file, I have warning" [WARNING] [launch_ros.actions.node]: Parameter file path is not a file: /home/nobata/lio-sam-pix-ws/install/lio_sam/share/lio_sam/config/params.yamal
[lio_sam_imuPreintegration-3] [WARN] [1721642911.117106600] [rcl.logging_rosout]: Publisher already registered for provided node name. If this is due to multiple nodes with the same name then all logs for that logger name will go out over the existing publisher. As soon as any node with that name is destructed it will unregister the publisher, preventing any further logs for that name from being published on the rosout topic.

I don't have error message regarding odom frame or with time. I sometimes get 'discarding message because the queue is full'
as I subscribe to the pointcloud topic in rviz2 and change frame from velo_link to other frames

Spirquel, 2024-07-22:
Okay, so actually your bag and your slam don't talk to each other. I can to explain to you the actual problem. When i see your rqt_graph, it show the node "/rosbag2_player" (it's your bag play) and the different node of lio-sam. so the connection between bag and slam doesn't exist. here a exemple with a connection beetwen a rosbag an slam : 
[image]
(sorry because i don't use your slam actually), we see the topic "/points_raw" (it's my topic for PointCloud2) and my slam is sub for this topic. So the actually problem for you is this link. The answer is in your warning "/home/nobata/lio-sam-pix-ws/install/lio_sam/share/lio_sam /config/params.yamal", you have a file config (the file for configuration the name of topic to you bag and other parameters) and this file are not the good format, you have a "params.yamal" but the real format is "params.yaml" so when you "colcon build --symlink-install" (--symlink-install is for change the python file and yaml files without to colcon build again) you build a bad params.yamal. For this you have to go to you config folder and change the format "yamal" to "yaml". it is with this file that your slam knows the topic to which sub, so without him the connection don't exist. You understand what i said ? (I'm French and my English is pretty bad). when you have this connection, the slam should react to it. you can see this connection thanks to rqt_graph.

wxyLuna2024, 2024-07-22:
Hi thank you for your speedy reply. I was able to spot this problem immediately after I posted this error and was about to edit my reply(I agree it is rather stupid!! I apologize for wasting your time. Thank you for your speedy reply!!) and I could see the GPS trajectory in my liosam. 

[image]
rqt_graph 
[image]

for |ros2 run tf2_tools view_frames it is showing this:
[image]

Spirquel, 2024-07-22:
t'inquietes. Actually your slam is sub only to you imu (when i read the msg for your rviz, your slam is maybe sub to you gps but i not sure.) The problem is maybe a param in your yaml, your slam said "point cloud ring channel not valid". So search in internet (web) your velodyne and number of this channel ring, it's at change here : 
[image]
usually it's 16 (for little lidar) or 32.

wxyLuna2024, 2024-07-23:
I have resolved the issue. It's due to the missing ring number that imageProjection.cpp was not working. Please refer to #452 . Both KITTI and KITTI360 dataset are missing ring number so it needs to be calculated manually. But some niche threshold needs further tuning as well, might as well refer to # #[https://blog.csdn.net/qq_39607707/article/details/124758082?spm=1001.2101.3001.6650.3&utm_medium=distribute.pc_relevant.none-task-blog-2~default~CTRLIST~Rate-3-124758082-blog-122576884.pc_relevant_3mothn_strategy_recovery&depth_1-utm_source=distribute.pc_relevant.none-task-blog-2~default~CTRLIST~Rate-3-124758082-blog-122576884.pc_relevant_3mothn_strategy_recovery&utm_relevant_index=3](url) if you are interested

L1MICH, 2025-09-09:
Hello everyone,

It looks like I've run into the same issue. After inspecting the code, I discovered that at runtime, the program uses the **params.yaml from install/lio_sam/share/lio_sam/config/params.yaml**, not the one from **src/LIO-SAM/config/params.yaml**.

This means any changes made to src/LIO-SAM/config/params.yaml won't take effect until the project is rebuilt. In my case, the odom → base_link transform, which needs to be published by the odometry node, was not being received correctly because the topic name in the effective params.yaml was wrong.

I was able to resolve this by **modifying src/LIO-SAM/config/params.yaml and then re-running colcon build**.

Hope this helps!