TixiaoShan/LIO-SAM #490: Point cloud timestamp not available, deskew function disabled, system will drift significantly!

Mubashir-alam, 2024-04-07:
I have compile `LIO-SAM` for ros noetic and running rosbag file `2011_09_30_drive_0028.bag` the map is incorrect I have swape `extrinsicRot` and `extrinsicRPY` in `config/params.yaml` file 
[image]
get the following warning 
```
 WARN] [1712523467.662934945]: Point cloud timestamp not available, deskew function disabled, system will drift significantly!
[ WARN] [1712523471.275059903]: Large velocity, reset IMU-preintegration!
```

Wuxinao, 2024-05-11:
hello, have you solved the problem?

Mubashir-alam, 2024-05-12:
Unfortunately No

RodrigofrBastos, 2024-07-18:
same problem running ROS2 branch

Petros626, 2024-09-28:
You must search in the old issues, I think this is already answered

daiyuuuuu, 2025-12-10:
@Petros626 excuse me could you please offer some details，I cant solve this problem

Petros626, 2025-12-10:
@daiyuuuuu solved in the old issues.