TixiaoShan/LIO-SAM #492: Increase Point Cloud Density in SLAM Mapping with unitree GO2 EDU Bot

Bharannn, 2024-04-30:
Hi there,

I hope this message finds you well. I've been using the unitree GO2 EDU bot equipped with the HESAI XT-16 LIDAR for SLAM mapping purposes. While performing SLAM and saving the map, I've observed that the point cloud density is lower than expected, even when utilizing the LIDAR at its highest RPM (1200 rpm) as per the documentation [here](https://support.unitree.com/home/en/developer/SLAM%20and%20Navigation_service).

I'm reaching out to inquire if there's a way to adjust the point cloud density directly through the SLAM process. I've reviewed the documentation provided but haven't found any specific instructions on how to address this issue.

Would it be possible for you to provide guidance on any configuration or parameter files that might allow me to adjust the point cloud density during SLAM mapping? Alternatively, if there are any other recommended solutions or approaches to increase the density of the point cloud output, I would greatly appreciate your insights.

Thank you for your attention to this matter, and I look forward to your assistance in resolving this issue.

XiaomingQu0913, 2024-05-14:
lio-sam only saves the plane points and edge points of the keyframe. If you want to save the full point cloud, you need to implement it yourself: 1. Read the implementation in ImageProjection::projectPointCloud() and save the undistorted points and their timestamps. 2.stich the point clouds in 1 according to the trajectory. lio-sam does not save every frame, you may need to interpolate.

XiaomingQu0913, 2024-05-14:
You'd better reassign a thread for file writing operations to avoid affecting real-time performance

arpitpara, 2024-05-24:
I've been checking the same, but it seem like no filters present in image_projection.cpp to downsample the data other than downsampling rate if present.
I feel it should be map_optimisation.cpp where all the registration of the cloud is happening. the points are filtered through leaf size. I reduced the leaf sizes to 0.001 and it gives me warning that data overload might happen.

``Leaf size is too small for the input dataset. Integer indices would overflow.[pcl::VoxelGrid::applyFilter] Leaf size is too small for the input dataset. Integer indices would overflow``

[image]

XiaomingQu0913, 2024-05-25:
> I've been checking the same, but it seem like no filters present in image_projection.cpp to downsample the data other than downsampling rate if present. I feel it should be map_optimisation.cpp where all the registration of the cloud is happening. the points are filtered through leaf size. I reduced the leaf sizes to 0.001 and it gives me warning that data overload might happen.
> 
> `Leaf size is too small for the input dataset. Integer indices would overflow.[pcl::VoxelGrid::applyFilter] Leaf size is too small for the input dataset. Integer indices would overflow` ![Screenshot from 2024-05-24 10-23-04](https://private-user-images.githubusercontent.com/58958058/333570272-28ed98b6-9419-48e8-9aad-70b7b3b64e8d.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MTY2MDMxOTksIm5iZiI6MTcxNjYwMjg5OSwicGF0aCI6Ii81ODk1ODA1OC8zMzM1NzAyNzItMjhlZDk4YjYtOTQxOS00OGU4LTlhYWQtNzBiN2IzYjY0ZThkLnBuZz9YLUFtei1BbGdvcml0aG09QVdTNC1ITUFDLVNIQTI1NiZYLUFtei1DcmVkZW50aWFsPUFLSUFWQ09EWUxTQTUzUFFLNFpBJTJGMjAyNDA1MjUlMkZ1cy1lYXN0LTElMkZzMyUyRmF3czRfcmVxdWVzdCZYLUFtei1EYXRlPTIwMjQwNTI1VDAyMDgxOVomWC1BbXotRXhwaXJlcz0zMDAmWC1BbXotU2lnbmF0dXJlPWUyNjZmMTZjNzI3ZjgzOThhZDEwNWU2ZmIxZjM3Y2U1MDg5MGU5MmQ3Y2E2MDExYWI5ZjJmNjc1MDAwNWQ4YzMmWC1BbXotU2lnbmVkSGVhZGVycz1ob3N0JmFjdG9yX2lkPTAma2V5X2lkPTAmcmVwb19pZD0wIn0.-2hOku5zAz9QSRhgrRzdJcVTvMp1MmVPOYAzzK6QY18)

map_optimization downsampling is only used to build a local map of icp and has no effect on the density of the map you need. This warning is because the index of voxel exceeds the maximum value of the int32 type. It is not recommended that you set leaf_size too small, which may cause loam to not obtain a good enough line or plane when performing point-line or point-surface matching.
The map you get from the source code is actually the feature points of the keyframes, so there may be two reasons for the sparse map points:
- Key frames are at least 1/2 of lidar frames, and may be sparser at slow speeds
- in feature extraction, points that are neither edge nor plane points are discarded

If you want to obtain a dense point cloud, it is not recommended that you obtain it from the source code. You may need to stitch it yourself based on the trajectory.

arpitpara, 2024-05-25:
Yeah Understood. Well Thanks for the explanation.
I am thinking of recording a PCAP file parallelly for the same time frame as mapping. I am getting a trajectory.txt or .pcd file that I can use further to stitch all the raw frames further through trajectory file.
How can I record a PCAP file for the same in the code itself. or If I record through tcpdump will it work?

Un1Lee, 2024-10-24:
> 嘿，你好
> 
> 我希望这条信息能找到你。我一直在使用配备 HESAI XT-16 激光雷达的 unitree GO2 EDU 机器人进行 SLAM 测绘。在执行 SLAM 和保存地图时，我观察到点云密度低于预期，即使按照[此处](https://support.unitree.com/home/en/developer/SLAM%20and%20Navigation_service)的文档以最高 RPM （1200 rpm） 使用 LIDAR 也是如此。
> 
> 我联系询问是否有办法直接通过 SLAM 过程调整点云密度。我已经查看了提供的文档，但没有找到有关如何解决此问题的任何具体说明。
> 
> 您是否可以提供有关任何配置或参数文件的指导，这些文件可能允许我在 SLAM 映射期间调整点云密度？或者，如果有任何其他推荐的解决方案或方法来增加点云输出的密度，我将非常感谢您的见解。
> 
> 感谢您对此事的关注，期待您的协助解决这一问题。

你好，我也在使用unitree的go2，我想请问下你怎么配置的param.yml。我现在启动了禾赛激光，读取了lowstate后根据时间新打了时间戳，再新发布imu的话题。我用的禾赛数据格式，然后x=-y，y=x。param.yml我设置了更多种lidar2imu的外参，但是似乎都有问题，都会漂。想请问下你怎么解决的？

gggzc, 2024-12-12:
> 你好，我也在使用unitree的go2，我想请问下你怎么配置的param.yml。我现在启动了禾赛激光，读取了lowstate后根据时间新打了时间戳，再新发布imu的话题。我用的禾赛数据格式，然后x=-y，y=x。param.yml我设置了更多种lidar2imu的外参，但是似乎都有问题，都会漂。想请问下你怎么解决的？

感觉可能是imu频率有点低，我录了rosbag看到狗自己imu频率可能只有50Hz

wilhelmsteve92, 2025-05-20:
@arpitpara were you able to create your own application that processes the raw point cloud data?

arpitpara, 2025-05-20:
No, I worked till setting up Lio-SLAM separately and getting the messages there. I was able to modify Lio-SLAM for Ros Noetic.
The Messages from GO2 EDU SDK is internally processed and used in LIO-SLAM that is inbuilt. For modifying the SLAM it is a bit complicated.
But I've found an Unofficial SDK for GO2 EDU, there work in the same field is pretty great and they've pretty smooth integration and scaling up with GO2 Robot.
Please let me know if I can help in something specifically more.

wilhelmsteve92, 2025-05-20:
@arpitpara I use the official Lio-sam-ros2 branch and set it up with a Hesai XT32.
So, you were able to get a better Pointcloud find your own noetic LIO-SAM?
Which SDK did you use? Couldn't find anything with LIO-SAM.

arpitpara, 2025-05-21:
No, I didn't get better point cloud quality.
There are multiple filters in LIO_SLAM that transforms point and takes a portion of it to calculate edges and surface more accurately.

arpitpara, 2025-08-31:
To my experience and knowledge, modifying anything inside default EDU system is next to impossible. The system is not having the raw code as it only has pre-built libraries. 
You can just use the system to get the data and use it inside some other application. I also recommend the same unofficial sdk for your use case too. This unofficial sdk has more flexibility of features and control over the raw code.