TixiaoShan/LIO-SAM #493: Large Map Closed Loop Issue

joewong00, 2024-05-21:
[image]

Hi, this is the map that is produced and we can clearly see that driving the same route is not able to close the loop in the map. There is no GPS correction for my data, may I know how can I tune the loop closure parameters to improve the map? Thanks

joewong00, 2024-05-24:
I have fixed the issue, turned out I set the wrong TF for the lidar to IMU. For those of you who are having this issue, you can use the formula here to help you to get the correct extrinsics parameters. And now my map looks great!

[image]

[image]

closed 2024-05-24 by joewong00

Zahra-Farajzadeh, 2024-07-03:
Hello, how did you calculate those matrix? what is theta?

joewong00, 2024-07-06:
> Hello, how did you calculate those matrix? what is theta?

Hi, theta is the angle transform from the lidar to IMU. If your IMU is placed 180 degree in the Z axis (yaw) from your lidar frame, then you can update the rotation matrix for R(z).