TixiaoShan/LIO-SAM #501: Large velocity, reset IMU-preintegration! error

liamm545, 2024-07-02:
Hello. 
I'm currently using lio-sam and have a question about it.

I am using a Velodyne VLP16 LiDAR and a 9-axis IMU sensor, but I'm encountering a "Large velocity, reset IMU-preintegration!" error, which is causing the map to become distorted.

[image]

So i checked the value of the velocity in "imuPreintegration.cpp" file.
 
`[ INFO] [1719903620.545044741]: Current velocity: [0.002522, 0.026950, -0.738660]
[ INFO] [1719903620.738969615]: Current velocity: [0.007577, 0.023546, -1.881389]
[ INFO] [1719903620.943298064]: Current velocity: [0.001869, 0.046400, -2.980822]
[ INFO] [1719903621.157010335]: Current velocity: [-0.118041, 0.058666, -4.721511]
[ INFO] [1719903621.362424426]: Current velocity: [-0.140791, 0.066850, -6.486963]
[ INFO] [1719903621.558399682]: Current velocity: [-0.101092, 0.076252, -8.518886]
[ INFO] [1719903621.761393458]: Current velocity: [0.281467, 0.217931, -10.953489]
[ INFO] [1719903621.952568893]: Current velocity: [1.122804, 0.120016, -12.950680]
[ INFO] [1719903622.158289919]: Current velocity: [1.680549, 0.042164, -14.219228]
[ INFO] [1719903622.372236656]: Current velocity: [3.331799, -0.166409, -15.497406]
[ INFO] [1719903622.554968747]: Current velocity: [3.902030, -0.279297, -16.066641]
[ INFO] [1719903622.773346387]: Current velocity: [4.908000, -0.509627, -16.566099]
[ INFO] [1719903622.968782143]: Current velocity: [5.554502, -0.687169, -17.043941]
[ INFO] [1719903623.163830851]: Current velocity: [6.021399, -0.821991, -17.636927]
[ INFO] [1719903623.373023411]: Current velocity: [6.214448, -0.916967, -18.270435]
[ INFO] [1719903623.583705434]: Current velocity: [6.334829, -1.178291, -18.548609]
[ INFO] [1719903623.775572965]: Current velocity: [6.279804, -1.355855, -19.073402]
[ INFO] [1719903623.967412059]: Current velocity: [6.397782, -1.531144, -19.617725]
[ INFO] [1719903624.168677612]: Current velocity: [6.314263, -1.626316, -20.175163]
[ INFO] [1719903624.391759498]: Current velocity: [6.021196, -1.745188, -20.933523]
[ INFO] [1719903624.583264917]: Current velocity: [5.880361, -1.908399, -21.742311]
[ INFO] [1719903624.780177006]: Current velocity: [5.683855, -2.025586, -22.464531]
[ INFO] [1719903624.993278509]: Current velocity: [5.238449, -2.003572, -23.489372]
[ INFO] [1719903625.196377777]: Current velocity: [5.091327, -2.009847, -24.491163]
[ INFO] [1719903625.386508596]: Current velocity: [5.210320, -2.105345, -25.452808]
[ INFO] [1719903625.591137373]: Current velocity: [5.523574, -2.261984, -26.401688]
[ INFO] [1719903625.788424844]: Current velocity: [5.387509, -2.299114, -27.190140]
[ INFO] [1719903625.989044756]: Current velocity: [5.658954, -2.202892, -28.250267]
[ INFO] [1719903626.193373597]: Current velocity: [5.735162, -2.201276, -28.921423]
[ INFO] [1719903626.403031391]: Current velocity: [5.670805, -2.154603, -29.934925]
[ WARN] [1719903626.403047604]: Large velocity, reset IMU-preintegration!
[ INFO] [1719903626.795157021]: Current velocity: [0.005990, -0.224255, -0.929813]
[ INFO] [1719903627.002490002]: Current velocity: [-0.974400, 0.665856, -2.595716]
[ INFO] [1719903627.215069760]: Current velocity: [-92.187363, -8.238041, -34.974682]
[ WARN] [1719903627.215083046]: Large velocity, reset IMU-preintegration!
[ INFO] [1719903627.598015582]: Current velocity: [0.090168, -0.061924, -1.036262]
[ INFO] [1719903627.814239665]: Current velocity: [0.464561, -0.358684, -2.591830]
[ INFO] [1719903628.004267369]: Current velocity: [71.447754, -26.926102, -63.163818]
[ WARN] [1719903628.004281328]: Large velocity, reset IMU-preintegration!`

As observed, I noticed significant spikes in velocity even when the vehicle was stationary. However, it seems there are no issues with the IMU sensor readings (attached below).

`IMU acc: 
x: 0.0909259, y: -0.136082, z: -0.985991
IMU gyro: 
x: 0, y: -0, z: 0
IMU roll pitch yaw: 
roll: -3.03875, pitch: -0.0671442, yaw: -1.52948

IMU acc: 
x: 0.0910207, y: -0.136188, z: -0.986981
IMU gyro: 
x: 0, y: -0, z: 0
IMU roll pitch yaw: 
roll: -3.03875, pitch: -0.0671442, yaw: -1.52948

IMU acc: 
x: 0.0910207, y: -0.136188, z: -0.986981
IMU gyro: 
x: 0, y: -0, z: 0
IMU roll pitch yaw: 
roll: -3.03875, pitch: -0.0671442, yaw: -1.52948

IMU acc: 
x: 0.0910207, y: -0.136188, z: -0.986981
IMU gyro: 
x: 0, y: -0, z: 0
IMU roll pitch yaw: 
roll: -3.03875, pitch: -0.0671442, yaw: -1.52948

^CIMU acc: 
x: 0.0911574, y: -0.137171, z: -0.986863
IMU gyro: 
x: 0, y: -0, z: 0
IMU roll pitch yaw: 
roll: -3.03894, pitch: -0.0671263, yaw: -1.52946

[ INFO] [1719905391.513121174]: Current velocity: [0.605741, 0.026891, -18.718386]
IMU acc: 
x: 0.0910207, y: -0.136188, z: -0.986981
IMU gyro: 
x: 0, y: -0, z: 0
IMU roll pitch yaw: 
roll: -3.03894, pitch: -0.0671263, yaw: -1.52946

IMU acc: 
x: 0.0910207, y: -0.136188, z: -0.986981
IMU gyro: 
x: 0, y: -0, z: 0
IMU roll pitch yaw: 
roll: -3.03894, pitch: -0.0671263, yaw: -1.52946

[lio_sam_rviz-8] killing on exit
IMU acc: 
x: 0.0910207, y: -0.136188, z: -0.986981
IMU gyro: 
x: 0, y: -0, z: 0
IMU roll pitch yaw: 
roll: -3.03894, pitch: -0.0671263, yaw: -1.52946
`

What could be the problem? I would appreciate your assistance. 
Thank you.

Zahra-Farajzadeh, 2024-07-03:
Hello I have the same problem with liosam
[image]
[image]

liamm545, 2024-07-06:
Hi, @Zahra-Farajzadeh 
I think wrong imu-lidar calibration is the reason.
When i change extrinsicRot by unit matirx, odometry looks a bit shaky, but the above issue has been resolved.

liamm545, 2024-08-29:
Hi, @Zahra-Farajzadeh.
I'm sorry for the late reply.

I think you should modify gpsCovThreshold and poseCovThreshold params.
Also you can check(debug) gps data in "mapOptimization.cpp" (Especially "addGPSFactor" function)

I'm sorry I couldn't be of any help.

gertrnn, 2024-12-03:
> This is my case.
> 
> If you look at this picture, you can see the difference between gps odometry and imu odometry. (The small axis is imu odometry, and the relatively large axis at the bottom is gps odometry.) ![Screenshot from 2024-08-29 17-35-33](https://private-user-images.githubusercontent.com/100026444/362599388-adf6c258-0206-4139-b8b3-b598fadb3035.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MzI1NTYyMTcsIm5iZiI6MTczMjU1NTkxNywicGF0aCI6Ii8xMDAwMjY0NDQvMzYyNTk5Mzg4LWFkZjZjMjU4LTAyMDYtNDEzOS1iOGIzLWI1OThmYWRiMzAzNS5wbmc_WC1BbXotQWxnb3JpdGhtPUFXUzQtSE1BQy1TSEEyNTYmWC1BbXotQ3JlZGVudGlhbD1BS0lBVkNPRFlMU0E1M1BRSzRaQSUyRjIwMjQxMTI1JTJGdXMtZWFzdC0xJTJGczMlMkZhd3M0X3JlcXVlc3QmWC1BbXotRGF0ZT0yMDI0MTEyNVQxNzMxNTdaJlgtQW16LUV4cGlyZXM9MzAwJlgtQW16LVNpZ25hdHVyZT1lYjcyMTk4YjZkYWYxYTljYzcwZjg5YmI4Mzk4ZGE1MzY3YTRkYzVhOGJjNWRlMDY0MTkwYWRhODQ2N2Q4YmQxJlgtQW16LVNpZ25lZEhlYWRlcnM9aG9zdCJ9.64gHWh5MvNq8K8z9fDjMxqUTkcpC5MT3tctc2Ct7c_U)
> 
> Although there are differences between IMU odometry and GPS odometry, during the map generation process, optimization is performed, which helps the odometries stay similar.
> 
> ![Screenshot from 2024-08-29 17-38-22](https://private-user-images.githubusercontent.com/100026444/362600122-cd7771fd-18d3-4e48-ab4c-683610a67948.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MzI1NTYyMTcsIm5iZiI6MTczMjU1NTkxNywicGF0aCI6Ii8xMDAwMjY0NDQvMzYyNjAwMTIyLWNkNzc3MWZkLTE4ZDMtNGU0OC1hYjRjLTY4MzYxMGE2Nzk0OC5wbmc_WC1BbXotQWxnb3JpdGhtPUFXUzQtSE1BQy1TSEEyNTYmWC1BbXotQ3JlZGVudGlhbD1BS0lBVkNPRFlMU0E1M1BRSzRaQSUyRjIwMjQxMTI1JTJGdXMtZWFzdC0xJTJGczMlMkZhd3M0X3JlcXVlc3QmWC1BbXotRGF0ZT0yMDI0MTEyNVQxNzMxNTdaJlgtQW16LUV4cGlyZXM9MzAwJlgtQW16LVNpZ25hdHVyZT0wNWU4YzViMDUwYmQyYjQxOTljMzE1N2NhNzk2MzY4NzJkMDI4NmRkMzlkYzlkMTA4MGY5OTc0YzI5NjczZDFkJlgtQW16LVNpZ25lZEhlYWRlcnM9aG9zdCJ9.vCIiMYGab55tugrHqFshPo8S8QbFI07pLVnDKm3tzbU) And although there were differences between IMU odometry and GPS odometry at first, you can see that the map is well-constructed as optimization and loop closure are eventually performed.
> 
> I recommend that you run it with SC-LIO-SAM.
> 
> Thanks.

Hello, I'm new to this field. I have a Hesai XT-16 LiDAR, and I'm trying to change the parameters and files. I can run the algorithm, but I can't get the map to save. Could you help me?

liamm545, 2024-12-04:
Hi, @gertrnn.
Please check “savePCD” param in params.yaml

closed 2024-12-05 by liamm545

gertrnn, 2024-12-05:
> Hola,@gertrnnPor favor , compruebe el parámetro “savePCD” en params.yaml

Sorry, what I mean is that it doesn't generate a map, the LiDAR works, but the LIO-SAM or map creation doesn't work

liamm545, 2024-12-05:
@gertrnn,
I would like to know in detail whether the issue lies with odometry not being published correctly, loop closing not being performed properly, or if there is another problem. 

It would be even better if you could provide some screenshots for clarification.