TixiaoShan/LIO-SAM #502: docker--could not initialize OpenGL for RasterGLSurface, reverting to RasterSurface.

wushijie777, 2024-07-06:
when i roslaunch lio_sam run.launch in docker, it could not open the rviz, and it shows
[image]

Zahra-Farajzadeh, 2024-07-09:
> when i roslaunch lio_sam run.launch in docker, it could not open the rviz, and it shows ![bug](https://private-user-images.githubusercontent.com/168325524/346254871-86b5b32a-86b5-4fee-a1d7-ac4232f17752.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MjA1MTU5MjYsIm5iZiI6MTcyMDUxNTYyNiwicGF0aCI6Ii8xNjgzMjU1MjQvMzQ2MjU0ODcxLTg2YjViMzJhLTg2YjUtNGZlZS1hMWQ3LWFjNDIzMmYxNzc1Mi5wbmc_WC1BbXotQWxnb3JpdGhtPUFXUzQtSE1BQy1TSEEyNTYmWC1BbXotQ3JlZGVudGlhbD1BS0lBVkNPRFlMU0E1M1BRSzRaQSUyRjIwMjQwNzA5JTJGdXMtZWFzdC0xJTJGczMlMkZhd3M0X3JlcXVlc3QmWC1BbXotRGF0ZT0yMDI0MDcwOVQwOTAwMjZaJlgtQW16LUV4cGlyZXM9MzAwJlgtQW16LVNpZ25hdHVyZT0yMjlkZmIzYmYyODA5NGY0MTAwOGU0M2VjYzFjZjUxMzZlMWI4ZTdlNTZlM2VmMWU0MDk2NzJkMjA1ODE4YWM5JlgtQW16LVNpZ25lZEhlYWRlcnM9aG9zdCZhY3Rvcl9pZD0wJmtleV9pZD0wJnJlcG9faWQ9MCJ9.NQvsycuanTdkg4azopYlO2iqTMqLjm-s5ouJk0U6DTg)

Hi, I had same problem when using docker. I used **XLaunch**  server to use rviz

bomena, 2024-07-24:
Hello, I had same problem.