TixiaoShan/LIO-SAM #512: How to deal with GPS error?

Anqfl, 2024-09-20:
When I use the GPS data of the dataset, the following error occurs. How should I solve it? Does anyone understand?
[ERROR] [1726843140.213914157]: Client [/lio_sam_mapOptmization] wants topic /ublox_gps/fix to have datatype/md5sum [nav_msgs/Odometry/cd5e73d190d741a2f92e81eda573aca7], but our version has [sensor_msgs/NavSatFix/2d3a8cd499b9b4a0249fb98fd05cfa48]. Dropping connection.
[ERROR] [1726843140.215444337]: Client [/ekf_gps] wants topic /ublox_gps/fix to have datatype/md5sum [nav_msgs/Odometry/cd5e73d190d741a2f92e81eda573aca7], but our version has [sensor_msgs/NavSatFix/2d3a8cd499b9b4a0249fb98fd05cfa48]. Dropping connection.

Anqfl, 2024-09-23:
I haved solved

karthikbolla, 2024-10-03:
@Anqfl Can you please mention how did you solved this issue ?