TixiaoShan/LIO-SAM #524: PCD with global coordinates?

antoniomontana1992, 2024-12-12:
Is it possible to create .pcd files with global coordinates?

stah-ba, 2025-01-30:
@antoniomontana1992 did you solve the issue? Or how did you continue?

lukasj-imar, 2025-01-31:
There is a field in the pcl object, which is called sensor_origin. You can use this field to define your offset in a geodetic reference frame.

antoniomontana1992, 2025-01-31:
I dont know if this is the right way, but what i did is to keep the first fixed reading from my RTK sensor, and then start liosam. While the map is finished, i make the conversion with pdal based on the first reading of my location.

stah-ba, 2025-01-31:
Alright, thanks for the quick answers! @lukasj-imar so you also use only the first GPS measurement as a offset as sensor origin for the mapped cloud?