TixiaoShan/LIO-SAM #525: 运行roslaunch lio_sam run.launch，rviz上面显示Global Status:Error ； base_link也是Status:Error

1508189250, 2025-01-06:
运行roslaunch lio_sam run.launch，rviz上面显示Global Status:Error；  base_link也是Status:Error；

控制台显示没有报错这是终端的打印。

rfr@fgfpr:~/ws_navigation$ roslaunch lio_sam run.launch
... logging to /home/jhl/.ros/log/f9c51cb0-cc10-11ef-bcdc-19832e929b3a/roslaunch-jhl-911pro-43349.log
Checking log directory for disk usage. This may take a while.
Press Ctrl-C to interrupt
Done checking log file disk usage. Usage is <1GB.

xacro: in-order processing became default in ROS Melodic. You can drop the option.
started roslaunch server http://jhl-911Pro:40465/

SUMMARY
========

PARAMETERS
 * /ekf_gps/base_link_frame: base_link
 * /ekf_gps/frequency: 50
 * /ekf_gps/imu0: imu_correct
 * /ekf_gps/imu0_config: [False, False, Fa...
 * /ekf_gps/imu0_differential: False
 * /ekf_gps/imu0_queue_size: 50
 * /ekf_gps/imu0_remove_gravitational_acceleration: True
 * /ekf_gps/map_frame: map
 * /ekf_gps/odom0: odometry/gps
 * /ekf_gps/odom0_config: [True, True, True...
 * /ekf_gps/odom0_differential: False
 * /ekf_gps/odom0_queue_size: 10
 * /ekf_gps/odom_frame: odom
 * /ekf_gps/process_noise_covariance: [1.0, 0, 0, 0, 0,...
 * /ekf_gps/publish_tf: False
 * /ekf_gps/sensor_timeout: 0.01
 * /ekf_gps/two_d_mode: False
 * /ekf_gps/world_frame: odom
 * /lio_sam/Horizon_SCAN: 1800
 * /lio_sam/N_SCAN: 16
 * /lio_sam/baselinkFrame: base_link
 * /lio_sam/downsampleRate: 1
 * /lio_sam/edgeFeatureMinValidNum: 10
 * /lio_sam/edgeThreshold: 1.0
 * /lio_sam/extrinsicRPY: [0, -1, 0, 1, 0, ...
 * /lio_sam/extrinsicRot: [-1, 0, 0, 0, 1, ...
 * /lio_sam/extrinsicTrans: [0.0, 0.0, 0.0]
 * /lio_sam/globalMapVisualizationLeafSize: 1.0
 * /lio_sam/globalMapVisualizationPoseDensity: 10.0
 * /lio_sam/globalMapVisualizationSearchRadius: 1000.0
 * /lio_sam/gpsCovThreshold: 2.0
 * /lio_sam/gpsTopic: odometry/gpsz
 * /lio_sam/historyKeyframeFitnessScore: 0.3
 * /lio_sam/historyKeyframeSearchNum: 25
 * /lio_sam/historyKeyframeSearchRadius: 15.0
 * /lio_sam/historyKeyframeSearchTimeDiff: 30.0
 * /lio_sam/imuAccBiasN: 6.435665935353257...
 * /lio_sam/imuAccNoise: 0.003993957088823881
 * /lio_sam/imuGravity: 9.80511
 * /lio_sam/imuGyrBiasN: 3.564031869636761...
 * /lio_sam/imuGyrNoise: 0.001563634394969...
 * /lio_sam/imuRPYWeight: 0.01
 * /lio_sam/imuTopic: imu_raw
 * /lio_sam/lidarFrame: base_link
 * /lio_sam/lidarMaxRange: 1000.0
 * /lio_sam/lidarMinRange: 1.0
 * /lio_sam/loopClosureEnableFlag: True
 * /lio_sam/loopClosureFrequency: 1.0
 * /lio_sam/mapFrame: map
 * /lio_sam/mappingCornerLeafSize: 0.2
 * /lio_sam/mappingProcessInterval: 0.15
 * /lio_sam/mappingSurfLeafSize: 0.4
 * /lio_sam/numberOfCores: 4
 * /lio_sam/odomTopic: odometry/imu
 * /lio_sam/odometryFrame: odom
 * /lio_sam/odometrySurfLeafSize: 0.4
 * /lio_sam/pointCloudTopic: points_raw
 * /lio_sam/poseCovThreshold: 25.0
 * /lio_sam/rotation_tollerance: 1000
 * /lio_sam/savePCD: True
 * /lio_sam/savePCDDirectory: /home/jhl/ws_3d_n...
 * /lio_sam/sensor: velodyne
 * /lio_sam/surfFeatureMinValidNum: 100
 * /lio_sam/surfThreshold: 0.1
 * /lio_sam/surroundingKeyframeDensity: 2.0
 * /lio_sam/surroundingKeyframeSearchRadius: 50.0
 * /lio_sam/surroundingKeyframeSize: 50
 * /lio_sam/surroundingkeyframeAddingAngleThreshold: 0.2
 * /lio_sam/surroundingkeyframeAddingDistThreshold: 1.0
 * /lio_sam/useGpsElevation: False
 * /lio_sam/useImuHeadingInitialization: True
 * /lio_sam/z_tollerance: 1000
 * /navsat/broadcast_utm_transform: False
 * /navsat/broadcast_utm_transform_as_parent_frame: False
 * /navsat/delay: 0.0
 * /navsat/frequency: 50
 * /navsat/magnetic_declination_radians: 0
 * /navsat/publish_filtered_gps: False
 * /navsat/wait_for_datum: False
 * /navsat/yaw_offset: 0
 * /navsat/zero_altitude: True
 * /robot_description: <?xml version="1....
 * /rosdistro: noetic
 * /rosversion: 1.17.0

NODES
  /
    ekf_gps (robot_localization/ekf_localization_node)
    lio_sam_featureExtraction (lio_sam/lio_sam_featureExtraction)
    lio_sam_imageProjection (lio_sam/lio_sam_imageProjection)
    lio_sam_imuPreintegration (lio_sam/lio_sam_imuPreintegration)
    lio_sam_mapOptmization (lio_sam/lio_sam_mapOptmization)
    lio_sam_rviz (rviz/rviz)
    navsat (robot_localization/navsat_transform_node)
    robot_state_publisher (robot_state_publisher/robot_state_publisher)

auto-starting new master
process[master]: started with pid [43374]
ROS_MASTER_URI=http://localhost:11311

setting /run_id to f9c51cb0-cc10-11ef-bcdc-19832e929b3a
process[rosout-1]: started with pid [43399]
started core service [/rosout]
process[lio_sam_imuPreintegration-2]: started with pid [43406]
process[lio_sam_imageProjection-3]: started with pid [43407]
process[lio_sam_featureExtraction-4]: started with pid [43408]
process[lio_sam_mapOptmization-5]: started with pid [43409]
process[robot_state_publisher-6]: started with pid [43424]
process[ekf_gps-7]: started with pid [43426]
process[navsat-8]: started with pid [43432]
process[lio_sam_rviz-9]: started with pid [43442]
[ INFO] [1736155876.316524358]: ----> Image Projection Started.
[ INFO] [1736155876.318632133]: ----> Feature Extraction Started.
[ INFO] [1736155876.374833685]: ----> Map Optimization Started.
[ INFO] [1736155876.377670690]: ----> IMU Preintegration Started.

closed 2025-01-06 by 1508189250

TheSeanParker, 2025-04-17:
> 环境：ubuntu20.04 + ros1（noetic）+ gtsam-4.0.0-alpha2 运行roslaunch lio_sam run.launch，rviz上面显示Global Status:Error； base_link也是Status:Error；
> 
> 控制台显示没有报错这是终端的打印。
> 
> rfr@fgfpr:~/ws_navigation$ roslaunch lio_sam run.launch ... logging to /home/jhl/.ros/log/f9c51cb0-cc10-11ef-bcdc-19832e929b3a/roslaunch-fgfpr-43349.log Checking log directory for disk usage. This may take a while. Press Ctrl-C to interrupt Done checking log file disk usage. Usage is <1GB.
> 
> xacro: in-order processing became default in ROS Melodic. You can drop the option. started roslaunch server http://fgfpr:40465/
> # SUMMARY
> 
> PARAMETERS
> 
>     * /ekf_gps/base_link_frame: base_link
> 
>     * /ekf_gps/frequency: 50
> 
>     * /ekf_gps/imu0: imu_correct
> 
>     * /ekf_gps/imu0_config: [False, False, Fa...
> 
>     * /ekf_gps/imu0_differential: False
> 
>     * /ekf_gps/imu0_queue_size: 50
> 
>     * /ekf_gps/imu0_remove_gravitational_acceleration: True
> 
>     * /ekf_gps/map_frame: map
> 
>     * /ekf_gps/odom0: odometry/gps
> 
>     * /ekf_gps/odom0_config: [True, True, True...
> 
>     * /ekf_gps/odom0_differential: False
> 
>     * /ekf_gps/odom0_queue_size: 10
> 
>     * /ekf_gps/odom_frame: odom
> 
>     * /ekf_gps/process_noise_covariance: [1.0, 0, 0, 0, 0,...
> 
>     * /ekf_gps/publish_tf: False
> 
>     * /ekf_gps/sensor_timeout: 0.01
> 
>     * /ekf_gps/two_d_mode: False
> 
>     * /ekf_gps/world_frame: odom
> 
>     * /lio_sam/Horizon_SCAN: 1800
> 
>     * /lio_sam/N_SCAN: 16
> 
>     * /lio_sam/baselinkFrame: base_link
> 
>     * /lio_sam/downsampleRate: 1
> 
>     * /lio_sam/edgeFeatureMinValidNum: 10
> 
>     * /lio_sam/edgeThreshold: 1.0
> 
>     * /lio_sam/extrinsicRPY: [0, -1, 0, 1, 0, ...
> 
>     * /lio_sam/extrinsicRot: [-1, 0, 0, 0, 1, ...
> 
>     * /lio_sam/extrinsicTrans: [0.0, 0.0, 0.0]
> 
>     * /lio_sam/globalMapVisualizationLeafSize: 1.0
> 
>     * /lio_sam/globalMapVisualizationPoseDensity: 10.0
> 
>     * /lio_sam/globalMapVisualizationSearchRadius: 1000.0
> 
>     * /lio_sam/gpsCovThreshold: 2.0
> 
>     * /lio_sam/gpsTopic: odometry/gpsz
> 
>     * /lio_sam/historyKeyframeFitnessScore: 0.3
> 
>     * /lio_sam/historyKeyframeSearchNum: 25
> 
>     * /lio_sam/historyKeyframeSearchRadius: 15.0
> 
>     * /lio_sam/historyKeyframeSearchTimeDiff: 30.0
> 
>     * /lio_sam/imuAccBiasN: 6.435665935353257...
> 
>     * /lio_sam/imuAccNoise: 0.003993957088823881
> 
>     * /lio_sam/imuGravity: 9.80511
> 
>     * /lio_sam/imuGyrBiasN: 3.564031869636761...
> 
>     * /lio_sam/imuGyrNoise: 0.001563634394969...
> 
>     * /lio_sam/imuRPYWeight: 0.01
> 
>     * /lio_sam/imuTopic: imu_raw
> 
>     * /lio_sam/lidarFrame: base_link
> 
>     * /lio_sam/lidarMaxRange: 1000.0
> 
>     * /lio_sam/lidarMinRange: 1.0
> 
>     * /lio_sam/loopClosureEnableFlag: True
> 
>     * /lio_sam/loopClosureFrequency: 1.0
> 
>     * /lio_sam/mapFrame: map
> 
>     * /lio_sam/mappingCornerLeafSize: 0.2
> 
>     * /lio_sam/mappingProcessInterval: 0.15
> 
>     * /lio_sam/mappingSurfLeafSize: 0.4
> 
>     * /lio_sam/numberOfCores: 4
> 
>     * /lio_sam/odomTopic: odometry/imu
> 
>     * /lio_sam/odometryFrame: odom
> 
>     * /lio_sam/odometrySurfLeafSize: 0.4
> 
>     * /lio_sam/pointCloudTopic: points_raw
> 
>     * /lio_sam/poseCovThreshold: 25.0
> 
>     * /lio_sam/rotation_tollerance: 1000
> 
>     * /lio_sam/savePCD: True
> 
>     * /lio_sam/savePCDDirectory: /home/jhl/ws_3d_n...
> 
>     * /lio_sam/sensor: velodyne
> 
>     * /lio_sam/surfFeatureMinValidNum: 100
> 
>     * /lio_sam/surfThreshold: 0.1
> 
>     * /lio_sam/surroundingKeyframeDensity: 2.0
> 
>     * /lio_sam/surroundingKeyframeSearchRadius: 50.0
> 
>     * /lio_sam/surroundingKeyframeSize: 50
> 
>     * /lio_sam/surroundingkeyframeAddingAngleThreshold: 0.2
> 
>     * /lio_sam/surroundingkeyframeAddingDistThreshold: 1.0
> 
>     * /lio_sam/useGpsElevation: False
> 
>     * /lio_sam/useImuHeadingInitialization: True
> 
>     * /lio_sam/z_tollerance: 1000
> 
>     * /navsat/broadcast_utm_transform: False
> 
>     * /navsat/broadcast_utm_transform_as_parent_frame: False
> 
>     * /navsat/delay: 0.0
> 
>     * /navsat/frequency: 50
> 
>     * /navsat/magnetic_declination_radians: 0
> 
>     * /navsat/publish_filtered_gps: False
> 
>     * /navsat/wait_for_datum: False
> 
>     * /navsat/yaw_offset: 0
> 
>     * /navsat/zero_altitude: True
> 
>     * /robot_description: <?xml version="1....
> 
>     * /rosdistro: noetic
> 
>     * /rosversion: 1.17.0
> 
> 
> NODES / ekf_gps (robot_localization/ekf_localization_node) lio_sam_featureExtraction (lio_sam/lio_sam_featureExtraction) lio_sam_imageProjection (lio_sam/lio_sam_imageProjection) lio_sam_imuPreintegration (lio_sam/lio_sam_imuPreintegration) lio_sam_mapOptmization (lio_sam/lio_sam_mapOptmization) lio_sam_rviz (rviz/rviz) navsat (robot_localization/navsat_transform_node) robot_state_publisher (robot_state_publisher/robot_state_publisher)
> 
> auto-starting new master process[master]: started with pid [43374] ROS_MASTER_URI=http://localhost:11311
> 
> setting /run_id to f9c51cb0-cc10-11ef-bcdc-19832e929b3a process[rosout-1]: started with pid [43399] started core service [/rosout] process[lio_sam_imuPreintegration-2]: started with pid [43406] process[lio_sam_imageProjection-3]: started with pid [43407] process[lio_sam_featureExtraction-4]: started with pid [43408] process[lio_sam_mapOptmization-5]: started with pid [43409] process[robot_state_publisher-6]: started with pid [43424] process[ekf_gps-7]: started with pid [43426] process[navsat-8]: started with pid [43432] process[lio_sam_rviz-9]: started with pid [43442] [ INFO] [1736155876.316524358]: ----> Image Projection Started. [ INFO] [1736155876.318632133]: ----> Feature Extraction Started. [ INFO] [1736155876.374833685]: ----> Map Optimization Started. [ INFO] [1736155876.377670690]: ----> IMU Preintegration Started.

后来是如何解决的呀？

TheSeanParker, 2025-04-18:
# 坐标系变换
## 提取GNSS坐标系
`rostopic echo /gps/fix | grep frame_id` 
frame_id: "velodyne"
## 提取LiDAR坐标系
`rostopic echo /points_raw | grep frame_id` 
frame_id: "navsat_link"
## 提取IMU坐标系
`rostopic echo /imu_correct | grep frame_id` 
 frame_id: "base_link"

REGATTE, 2025-05-16:
hey, the link isnt working