TixiaoShan/LIO-SAM #529: When running with the Livox Mid360, the map saving process does not complete.

falconkun, 2025-01-21:
I want to create a map using the Livox Mid360, but when I execute the map saving command, it doesn't complete.
I've modified the settings in params.yaml and rviz2.rviz, but are there any other settings I should configure?
The Ouster OS1 worked without any issues.

### params.yaml
```
    # Topics
    pointCloudTopic: "/livox/lidar"                   # Point cloud data
    imuTopic: "/livox/imu"                        # IMU data
    odomTopic: "odometry/imu"                    # IMU pre-preintegration odometry, same frequency as IMU
    gpsTopic: "odometry/gpsz"                    # GPS odometry topic from navsat, see module_navsat.launch file

    # Frames
    lidarFrame: "livox_frame"
    baselinkFrame: "base_link"
    odometryFrame: "odom"
    mapFrame: "map"

    # GPS Settings
    useImuHeadingInitialization: false           # if using GPS data, set to "true"
    useGpsElevation: false                       # if GPS elevation is bad, set to "false"
    gpsCovThreshold: 2.0                         # m^2, threshold for using GPS data
    poseCovThreshold: 25.0                       # m^2, threshold for using GPS data

    # Export settings
    savePCD: false                               # https://github.com/TixiaoShan/LIO-SAM/issues/3
    savePCDDirectory: "/Downloads/LOAM/"         # in your home folder, starts and ends with "/". Warning: the code deletes "LOAM" folder then recreates it. See "mapOptimization" for implementation

    # Sensor Settings
    sensor: livox                               # lidar sensor type, either 'velodyne', 'ouster' or 'livox'
    N_SCAN: 4                                   # number of lidar channels (i.e., Velodyne/Ouster: 16, 32, 64, 128, Livox Horizon: 6)
    Horizon_SCAN: 6000                            # lidar horizontal resolution (Velodyne:1800, Ouster:512,1024,2048, Livox Horizon: 4000)
    downsampleRate: 1                            # default: 1. Downsample your data if too many
    # points. i.e., 16 = 64 / 4, 16 = 16 / 1
    lidarMinRange: 1.0                           # default: 1.0, minimum lidar range to be used
    lidarMaxRange: 1000.0                        # default: 1000.0, maximum lidar range to be used
```

### rviz2.rviz
```
      Topic:
        Depth: 5
        Durability Policy: Volatile
        History Policy: Keep Last
        Reliability Policy: System Default
        Value: /livox/lidar
      Use Fixed Frame: true
      Use rainbow: true
      Value: true
  Enabled: true
  Global Options:
    Background Color: 48; 48; 48
    Fixed Frame: base_link
    Frame Rate: 30
```

[image]

Joaquin-Navas, 2025-02-18:
Hi,

could you share how did you adapt LIO-SAM to MID360?

I've got MID360 working properly with FAST LIO ROS 2, but couldn't make it work in this case :(

Thanks

jonathantyl97, 2025-02-23:
@Joaquin-Navas 

Hey, I'm trying to use FAST-LIO ROS2 with mid360 also, I can scan with point cloud generated but there were no mapping and I couldn't save the map at all

Joaquin-Navas, 2025-02-25:
> [@Joaquin-Navas](https://github.com/Joaquin-Navas)
> 
> Hey, I'm trying to use FAST-LIO ROS2 with mid360 also, I can scan with point cloud generated but there were no mapping and I couldn't save the map at all.
> 
> May I know how did you do it?
> 
> Thanks!

What do you mean with _point cloud generated but there were no mapping_? and that you _couldn´t save the map at all_?

Are you referring you can SLAM and see the mapping in rviz but that the point cloud is not saved?

jonathantyl97, 2025-02-25:
@Joaquin-Navas 

https://drive.google.com/file/d/1Cwp29qvHg9128-An2TyJkhR_1vaoUy4h/view?usp=drive_link

I am facing such issue. I can have pointclouds, but there is no SLAM

Joaquin-Navas, 2025-02-25:
> [@Joaquin-Navas](https://github.com/Joaquin-Navas)
> 
> https://drive.google.com/file/d/1Cwp29qvHg9128-An2TyJkhR_1vaoUy4h/view?usp=drive_link
> 
> I am facing such issue. I can have pointclouds, but there is no SLAM

Please, open an issue in FAST LIO ROS2 repository specifying what command do you launch and mention me.

lopf35, 2025-07-13:
@falconkun I'm also trying to get LIO-SAM working with the mid360 Would be great to know how you did it.